lvgl/lvgl #591: [question] button state when move some distance

fallstool, 2018-11-29:
Hi, 

Now current version, when you pressed on the button and move some distance, the button state will not changed util you release the button. so it may cause some problem, when in the tabview,(or page ) that can scroll container. If I display a button full of the tabview in the section one, when I press the button and move, the tabview do not move to the second section.
I saw in the smartphone, when press the button, and move some distance the button state will changed to unpressed state autoly， and then can move to next page.
Is there any way in lvgl to acheive the same effect?

kisvegabor, 2018-11-29:
Hi,

You can use `lv_page_glue_obj(btn, true)` but it will allow to scroll the page and not to change between the pages.

But with a little hack you can achive it:
```c

/*Original the signal functions of a button*/
static lv_signal_func_t btn_signal;

some_function()
{
   ....
    lv_obj_t * btn = lv_btn_create(tab, NULL);
    lv_page_glue_obj(btn, true);

   /*Save the original button signal and set new one*/
    btn_signal = lv_obj_get_signal_func(btn);
    lv_obj_set_signal_func(btn, new_btn_signal);
   ....
 }

/*New signal handler for the button*/
static lv_res_t new_btn_signal(lv_obj_t * btn, lv_signal_t sign, void * param)
{
    lv_res_t res;

    /* Include the original button signal function */
    res = btn_signal(btn, sign, param);
    if(res != LV_RES_OK) return res;
    
    /*Call the scrollable's signal function*/
    lv_obj_t * tab_scrl = lv_obj_get_parent(btn);
    tab_scrl->signal_func(tab_scrl, sign, param);

    return res;
}

```

fallstool, 2018-11-29:
Hi, @kisvegabor thank you for your reply. I will try it with your suggestion.

embeddedt, 2018-12-23:
There has been no activity for 1 month so I will assume the issue is resolved. If you are still having problems feel free to reopen the issue and comment.

closed 2018-12-23 by embeddedt