lvgl/lvgl #594: Use circular buffer for lv_chart to increase efficiency

stheil, 2018-11-29:
Hi,

I was playing around with a chart object and noticed it moves the contents of the entire data array when adding a new point, which is a lot of overhead if you're displaying a lot of data points (around 1000 in my case). I modified the lv_chart object to use a circular buffer with one data pointer, which makes adding a data point take a constant amount of time, independent of the array size. Unfortunately this has the drawback of complicating changes to the array size, but I think that is an operation which is rarely done, definitely less often than adding a data point.

If that sounds interesting to you I'd be happy to make a pull request.

kisvegabor, 2018-11-30:
Hi,

Sounds well! 
Yes, please, create a Pull request.

embeddedt, 2018-12-10:
As the PR is merged I will close this issue. If you have questions or continue to have problems feel free to reopen it.

closed 2018-12-10 by embeddedt