lvgl/lvgl #596: Controls are always unable to change colors

longdadouzi, 2018-11-30:
Call the functions of the document, such as these:
lv_obj_t * par = lv_obj_create(lv_scr_act(), NULL); 
lv_obj_set_size(par, 100, 80); /*Set the size of the pa
lv_obj_t * obj1 = lv_obj_create(par, NULL); /*Create an object on th
lv_obj_set_pos(obj1, 10, 10); /*Set the position of th
He creates only one of the classes, and the parent class is displayed only when a subclass is deleted;
You can change the color normally here.

But when I use other components, such as sliding components and buttons, I can't make their colors appear, just wireframes.

longdadouzi, 2018-12-03:
But I set the style for the control. I call the function to get the theme and set the style for the control. You mean that he is blocked by the parent class. In addition, I don’t have a touch screen, can I provide some ideas for using buttons?

kisvegabor, 2018-12-03:
Please send a code what we can test and a screenshot. 
Do you use the [PC simulator](http://doc.littlevgl.com/#PC-simulator)? It would be good to use because this we can exclude hardware related things. 

About button usage see: http://doc.littlevgl.com/#Porting

longdadouzi, 2018-12-03:
I have read the documentation you said, or do not quite understand the input device, according to how to distinguish how the button's key value is associated with the controls on the screen.
The computer-side simulation was not used, and it seems that I have not found it.
I used demo_create() today; I can display a small number of colors, but the layout of the fonts is strange.
As shown：
[https://github.com/longdadouzi/question/blob/master/reply1.jpg](url)

longdadouzi, 2018-12-03:
As shown： [https://raw.githubusercontent.com/longdadouzi/question/master/reply1.jpg](url)

longdadouzi, 2018-12-03:
https://raw.githubusercontent.com/longdadouzi/question/master/reply1.jpg

longdadouzi, 2018-12-04:
My screen is 320*240 LCD. It is RGB565, I have configured it for 16 bits. The function is as follows:
Void tft_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
   U16 height,width;
   U16 i,j;
   Width=x2-x1+1; //get the width of the fill
   Height=y2-y1+1; //height
   For(i=0;i<height;i++)
   {
     LCD_SetCursor(x1, y1+i); //Set the cursor position
     IOB_LCD_CMD=DG_Lcd.WRamCmd;//Start writing RAM
     For(j=0;j<width;j++)
     {
       IOB_LCD_DATA=color_p->full>>8;//write data
       IOB_LCD_DATA=color_p->full;
       Color_p++;
     }
   }
   Lv_flush_ready();
}
This may happen because of what it is.

kisvegabor, 2018-12-04:
It seems correct. Have you verified that your driver really works well? You can test with something like this:
```c
    lv_color_t c[1000];
    uint16_t i;
    for(i = 0; i < 1000; i++) {
        c[i] = LV_COLOR_RED;
    }

    tft_flush(10, 20, 50, 80, c);
```

longdadouzi, 2018-12-04:
Yes, I didn't use littevGL before, it didn't matter. I think the problem will not be in my configuration. I changed the size of the configured screen and put it on another board. He can 800*480 is displayed on the screen.

longdadouzi, 2018-12-05:
I solved the previous problem, this is a driver issue. But now it has other problems. Flashing. It seems that the buffer is not enough. I don't know where I can configure, can I help me？
As shown:
[https://github.com/longdadouzi/question/blob/master/reply2.jpg](url)
https://github.com/longdadouzi/question/blob/master/reply2.jpg

kisvegabor, 2018-12-05:
Your link was not working so I edit your post to fix it. The required format is: [text](http://...) or just copy-paste a link.

You can set the buffer size in `lv_conf.h` with `LV_VDB_SIZE` however I'm almost sure it's not because of it. Have you tested the code snippet I posted above?

longdadouzi, 2018-12-06:
I did according to your test and successfully solved the problem. This is a problem with my screen driver, and lv_conf.h LV_VDB_SIZE cannot be set too large and cannot be set too small. My screen is 320 * 240, I set 320, LV_DPI I set it to 50, but I don't understand the role of LV_DPI
Thank you for your answer.

longdadouzi, 2018-12-06:
When I use controls, I can't use the Line Meter as I said in the documentation. I can't let him show the percentage as shown. I haven't found a document on how to display the percentage.
Thank you for your answer.
[image]

embeddedt, 2018-12-06:
This is the line meter documentation:

https://doc.littlevgl.com/#Line-meter

At the bottom there is an example showing how the percentage is displayed.

kisvegabor, 2018-12-06:
@longdadouzi 
It seems the original issue is solved so I close it to keep the things clear. 
If you have further questions about the line meter or anything else please open a dedicated issue for them.

closed 2018-12-06 by kisvegabor