lvgl/lvgl #598: STM32F429 Example project does not compile

PatrickV82, 2018-11-30:
I am trying to run the example project for the STM32F429 discovery board, that can found at https://github.com/littlevgl/lv_projects/tree/master/stm32f429_discovery_no_os

when i use the steps as described 

> Clone the repository somewhere: git clone https://github.com/littlevgl/lv_boards.git
 Step into the repository: cd lv_boards
Step into the folder of the desired project: cd folder_of_a_project
Initalize the submodules: git submodule update --init .
Open the project or copy/paste it where you want (e.g. IDE workspace)`

When I try to compile the project with atollic TrueSTUDIO
I get multiple file not found errors
>  fatal error: lvgl/lv_draw/lv_draw_img.h: No such file or directory

If I try to update the lvgl to the latest version instead of the one used in the example 
I also get compile errors 

> \hal_stm_lvgl\tft/tft.c:337:  undefined reference to `lv_color_to24'

Is there a working example of the project for the STM32F429 discovery ???

kisvegabor, 2018-11-30:
Hi,

That project is for AC6 SystemWorkbench (SW4STM32). To use it with Atollic you need to modify some configs in the project properties.

However is your only issue is `lv_color_to24` just change it to `lv_color_to32`

PatrickV82, 2018-12-03:
The project settings were not the problem, Atollic can import SW4STM32 projects.

after changing the lv_color_to24  to lv_color_to32 it compiled and worked

kisvegabor, 2018-12-03:
Great! :)

closed 2018-12-03 by kisvegabor