lvgl/lvgl #603: lv_imgbtn_set_src image

MShahrukh0, 2018-12-03:
how to change imgbtn image when focused using pc keyboard?

kisvegabor, 2018-12-03:
There is no simple way to do it right now. However, we are working on a new action system here: #316 
Probably there will be only one action per object with an action type parameter, for example: `LV_ACTION_CLICK`, `LV_ACTION_PRSSSED` etc.

I think we can have `LV_ACTION_FOCUS` and `LV_ACTION_DEFOCUS` too to allow changing the image in your action function. What do you think @embeddedt and @amirgon?

Until this feature is implemented you can extend the signal function of the image button like this:

```c
static lv_signal_func_t old_signal;

void func(void)
{
    lv_obj_t * imgbtn = lv_imgbtn_create(lv_scr_act(), NULL);
    old_signal = lv_obj_get_signal_func(imgbtn);
    lv_obj_set_signal_func(imgbtn, new_signal);
}

static lv_res_t new_signal(lv_obj_t * imgbtn, lv_signal_t sign, void * param)
{
    lv_res_t res;

    /* Include the ancient signal function */
    res = old_sign(imgbtn, sign, param);
    if(res != LV_RES_OK) return res;

    if(sign == LV_SIGNAL_FOCUS) {
        lv_imgbtn_set_src(imgbtn, LV_BTN_STATE_REL, &img_focused);
    }
    else if(sign == LV_SIGNAL_DEFOCUS) {
        lv_imgbtn_set_src(imgbtn, LV_BTN_STATE_REL, &img_defocused);
    }
    
    return res;
}

```

MShahrukh0, 2018-12-04:
thank you, I think this may solve my problem for now , but let me test this first then i'll let you know the result.

MShahrukh0, 2018-12-04:
It worked fine thank you.

closed 2018-12-04 by MShahrukh0

MShahrukh0, 2018-12-04:
One more issue is that when i set callback on LV_BTN_ACTION_PR and pressing LV_GROUP_KEY_ENTER then callback fuction is not activating , but if change action to CLICK from PR tthan it works fine,
Can you tell me why I am unable to use PR action when enter key is pressed?

kisvegabor, 2018-12-05:
Yes, it triggers the CLICK action. You can also use the LONG_PRESS action.
PRESS action was not added, but I've just added it to `dev-5.3` branch. Does it work well?

kisvegabor, 2018-12-05:
I just saw that you opened a new issue for this question. Let's continue the discussion there

MShahrukh0, 2018-12-05:
I haven't testes dev-5.3 , but is long_press works same as pressed?

MShahrukh0, 2018-12-05:
Ok I'll check this long_press and will reply you there on open issue