lvgl/lvgl #605: lv_imgbtn_set_action LV_BTN_ACTION_PR not working

MShahrukh0, 2018-12-04:
I am using pc keyboard and created a lv_imgbtn , if I set callback on click action and press Enter key then callback is working fine but when i set action to pressed and pressing enter key then its not working.
Can anyone help me in this?
Also I am using keyboars keys in pressed state , following is my keyboard key state code,

if(in.type == EV_KEY)
 {
            printf("key code %d\n",in.code);    
            if(in.code == BTN_MOUSE || in.code == BTN_TOUCH) 
            {
 	        if(in.value == 0)
       		{
                    evdev_button = LV_INDEV_STATE_REL;
                }       
                else if(in.value == 1)
                {
                    evdev_button = LV_INDEV_STATE_PR;
            	}
            }
	    else 
            {
 		//keyboard key
              	if(in.value == 0)
                  {
                     data->state = LV_INDEV_STATE_REL;
                  }
                else if(in.value == 1)
                 {
                     data->state = LV_INDEV_STATE_PR;
                     data->key = keycode_to_ascii(in.code);
	   	}
		return false;
	   }
			
 }

kisvegabor, 2018-12-05:
It triggers the CLICK action. You can also use the LONG_PRESS action.
PRESS action was not added, but I've just added it to dev-5.3 branch. Does it work well?

kisvegabor, 2018-12-05:
In your comments please use the 
` ```c `
`code`
`code`
`...`
` ``` `

to add sytax highlight to your code.

MShahrukh0, 2018-12-05:
LONG_PR also not working if I am presseing enter key

kisvegabor, 2018-12-05:
LONG_PR should work. Have you already set up the PC simulator? A test in the simulator would be great to be sure the library works, and the issue is in the driver.

MShahrukh0, 2018-12-05:
may be the issue is different what I did , I created three imgbtn and added them to the group but initially there is home screen where I have only labels , so when i pressed enter key it hide all the labels and unhide all the imgbtns so instead of showing just three buttons , the first button which by default focused in group get also clicked by that enter key and callback function called.
correct me if my explanation is wrong.

MShahrukh0, 2018-12-05:
No I am using on resberry pi like board , and all the function key which i assigned works fine , even enter key working as I want it to be except in this imgbtn case.
Can you tell me another way to use CLICK action as its work when enter key press and then released , here in my above mentioned code I am using only pressed mode of keyboard keys.

kisvegabor, 2018-12-05:
I see, but please make a test only with a singe imgbtn to see if the object itself is working or the issue is somewhere else.

MShahrukh0, 2018-12-05:
its working when I first pressed the enter key to display all three btn and hide labels, it directly call the callback function of first imgbtn then I exit from that function and after that it works fine I press enter callback function called . The Only issue is when I press first time then it should not click the button instead wait for me to press the enter key again.

kisvegabor, 2018-12-05:
I see. Maybe set the `action` function of the imgbtn when you unhide it. When you hied it set the `actions` to `NULL`

MShahrukh0, 2018-12-05:
still the same issue.

MShahrukh0, 2018-12-05:
I solve my problem with a different logic.
I now don't return LV_GROUP_KEY_ENTER when I am going in main menu.
Thanks for all your time and support.

closed 2018-12-05 by MShahrukh0