lvgl/lvgl #606: can we add new symbols?

MShahrukh0, 2018-12-04:
I have created a list and want to add few buttons in it , I can add button but I need different symbols such SYMBOL_USER, 
Can anyone tell me how can I add symbol?

kisvegabor, 2018-12-05:
Hi,

Built-in symbols are special, but you can add your own symbols like any other font. I suggest using the `Font Awesome` font (you can download it from the font coverter's page too). Set the range and the required symbol's HTML code in the list (text area) on the website.

After that, you can join your font to a built-in symbol font: `lv_font_add(&my_symbol_font_20, &lv_font_symbol_20);`

MShahrukh0, 2018-12-05:
Ok will try this and report you, but my major issue is which i asked in another thread.

MShahrukh0, 2018-12-05:
the symbol font which I downlaoded from font converter's website doesn't contain the symbols which I need, so tell me if I download one svg format icon from fontawesome website then how can I use it in my code?

kisvegabor, 2018-12-06:
If you have the icon as an image then you can use the [image converter tool](https://littlevgl.com/image-to-c-array) and add the result image to the list. I suggest converting the icon to `png` format
```c
LV_IMG_DECLARE(my_icon);
lv_list_add(list, &my_icon, "Text", action);
```

MShahrukh0, 2018-12-07:
that's I already did. Thanks for your support

closed 2018-12-07 by MShahrukh0