lvgl/lvgl #608: How to update interface data in real time?

as564335sa, 2018-12-06:
Hi:
 How can i update the interface data in real time? I created a task for a second, but the data couldn't change in time，Do you have any good idea?

embeddedt, 2018-12-06:
@as564335sa 

Please explain your issue in more detail. I'm not sure what you're trying to accomplish. If you mean changing the LittlevGL data in more than one thread, you would need one thread to call lv_task_handler repeatedly (i.e. every 10 ms) and one thread to change the data.

> If you need to use other task or threads you need one mutex which should be taken before calling lv_task_handler and released after it. In addition, you have to use to that mutex in other tasks and threads around every LittlevGL (lv_...) related code. This way you can use LittlevGL in a real multitasking environment. Just use a mutex to avoid concurrent calling of LittlevGL functions.

closed 2018-12-06 by as564335sa

effeci, 2018-12-09:
what's this mutex ?

embeddedt, 2018-12-09:
@effeci You have to create it using your OS and make sure that you lock it and unlock it around LittlevGL related code.

kisvegabor, 2018-12-09:
The Mutexes are a basic element of every OS. You can Google on it to learn how it (and Operating systems in general) works.

teaalltr, 2024-01-26:
Could you provide an example of this?