lvgl/lvgl #609: How can I to set the font size?

tianyoung, 2018-12-07:


fallstool, 2018-12-07:
Hi，
https://github.com/littlevgl/doc/blob/master/Fonts.md

tianyoung, 2018-12-07:
All right, I see. Thanks~. This method is not elegant enough maybe.

embeddedt, 2018-12-07:
@tianyoung Why isn't this method elegant?

If you want dynamic font size support, generate more static fonts using https://littlevgl.com/ttf-font-to-c-array and choose between them at runtime. TTF/vector fonts are not currently supported.

tianyoung, 2018-12-10:
Thank you so much.  @embeddedt 
I want to ask more.
Where are simplified chinese font without license?

BesitzeRuf, 2018-12-10:
@tianyoung yes, additional functionality should be implemented...
1) **lv_font_t** structure has a number of pointers to const,
2) There is no function for deleting font. Maybe it can be a feature request for 6.0 version

kisvegabor, 2018-12-10:
@tianyoung You should initialize an `lv_font_t` variable.

I created a quick example:
```c
const uint8_t * my_bitmap_get(const lv_font_t * f,uint32_t unicode)
{
    memset(bitmap_buf, 0, sizeof(bitmap_buf));

    int16_t w = f->get_width(f, unicode);
    uint8_t x;
    uint8_t y;

    /*Draw a stripped bitmap*/
    for(y = 0; y < f->h_px; y+=2) {

        for(x = 0; x < w; x++) {
            bitmap_buf[y * w + x] = 0xFF;
            printf("set: %d\n", y * w + x);
        }
    }

    return bitmap_buf;

}

int16_t my_width_get(const lv_font_t * f,uint32_t unicode)
{
    /*Dummy example: '1' -> 1 px wide, '2' -> 2 px wide etc.*/
    if(unicode >= '0' && unicode <= '9') {
        return unicode - '0';  /*0..9*/
    }
    return -1;
}

...

    /*Initialize the font*/
    static lv_font_t dummy_font;
    dummy_font.get_bitmap = my_bitmap_get;
    dummy_font.get_width = my_width_get;
    dummy_font.h_px = 20;
    dummy_font.unicode_first = 0x30;  /*'0'*/
    dummy_font.unicode_last = 0x39;  /*'9'*/
    dummy_font.monospace = 0;
    dummy_font.unicode_list = NULL;
    dummy_font.bpp = 8;           /*8 bit per pixel for simplicity. 1 byte -> 1 pixel*/

    /*Used by "normal" converted fonts. Just set them to zero*/
    dummy_font.glyph_cnt = 0;
    dummy_font.glyph_dsc = NULL;

   /*Create a style with the font*/
    static lv_style_t font_test_style;
    lv_style_copy(&font_test_style, &lv_style_plain);
    font_test_style.text.font = &dummy_font;

    /*Create a label with the style*/
    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_text(label, "123456789");
    lv_label_set_style(label, &font_test_style);
```

Result:
[image]

kisvegabor, 2018-12-10:
@BesitzeRuf 
> There is no function for deleting font. Maybe it can be a feature request for 6.0 version

How do you mean "deleting font"? Fonts are just global (or static) `lv_font_t` structures. They are alloacted in compile time. If the user alloactes them dynamically then he should free them too.

BesitzeRuf, 2018-12-11:
@kisvegabor With FreeType library we can create raster from vector fonts dynamically in run-time, but it  requires extra memory to keep generated font bitmaps (in ram or ext. ram).

There is a function for adding a font as child to parent font (sorry, here I don't clearly understand the meaning of "charter set", if shouldn't it be a "character set"? )
void lv_font_add(lv_font_t * child, lv_font_t * parent), but no function to remove (all or just child) dynamically created font. Of-course user should free related things... I just mean, that it is not implemented in current version.

BesitzeRuf, 2018-12-12:
@kisvegabor  thank you too :)

kisvegabor, 2018-12-13:
@BesitzeRuf 
I added the `lv_font_remove` too: https://github.com/littlevgl/lvgl/commit/dea81daa79c57a261a7d7e61d5b078e42921d340