lvgl/lvgl #610: How to customise keyboard keys?

MShahrukh0, 2018-12-07:
I want to customise keyboard keys , for example if I press KEY_1 then KEY_2 return value is correctly returned but in textbox space typed in place of 2.
```
evdev_init();

	/*evdev to handle kb input*/
	lv_indev_drv_t kb_drv;
	lv_indev_drv_init(&kb_drv);
	kb_drv.type = LV_INDEV_TYPE_KEYPAD;
	kb_drv.read = evdev_read;

	/*assign kb to group*/
	kb_indev = lv_indev_drv_register(&kb_drv);
	lv_indev_set_group(kb_indev, group2);
```
here I am using pc keyboard.

embeddedt, 2018-12-08:
@MShahrukh0 I don't understand exactly what the problem is. Can you explain further?

kisvegabor, 2018-12-08:
Please share your `evdev_read` function to see what you have right now.

MShahrukh0, 2018-12-10:
```
if(in.value == 0){
                     data->state = LV_INDEV_STATE_REL;
                     //data->key = keycode_to_ascii(in.code);
                    // printf("Released key code %d\n",in.code);
                    }
                else if(in.value == 1){
                     data->state = LV_INDEV_STATE_PR;
                     data->key = keycode_to_ascii(in.code);
                     printf("Pressed key code %d, returned key code %d\n",in.code,data->key);
                    
	   	}
``
here I am using keyboard key in pressed state.
I have seen that value key value 9 is assignd as LV_GROUP_KEY_NEXT but actually it is the value of KEY_8 in keyboard, so even if I define LV_GROUP_KEY_NEXT to another value then still key value 9 functioning as LV_GROUP_KEY_NEXT, however LV_GROUP_KEY_NEXT gives the value which we defined.
So I just want to know how to change such keys and resassign them according to my need?

kisvegabor, 2018-12-10:
I see. `LV_GROUP_KEY_...` defines are used consequently so you can change them in `lv_group.h` to any value.  For example `LV_GROUP_KEY_NEXT     0xFFFFFF01`

Please test it. If is works, probably these defines should be updated in the library.

MShahrukh0, 2018-12-11:
That's what i already did, I changed the value to different key and this LV_GROUP_KEY_NEXT start giiving that assigned value but when I pressed the previously assigned key than still its works as LV_GROUP_KEY_NEXT.

kisvegabor, 2018-12-12:
I've tested it in the PC simulator with the `keyboard` driver. I added
```c
        case SDLK_TAB:
            return LV_GROUP_KEY_NEXT;
```
to `keycode_to_ascii()` function and chaged `LV_GROUP_KEY_NEXT 0xFFFFFF09`.

And it worked for me.

MShahrukh0, 2018-12-13:
I have used my logics to solve my issue , thanks for all the supports .

closed 2018-12-13 by MShahrukh0

kisvegabor, 2018-12-13:
So finally `LV_GROUP_KEY_NEXT 0xFFFFFF09` worked?