lvgl/lvgl #611: screen display stripes

relliktei, 2018-12-07:
I am using RT1052 MCU and 480x854 RGB lcd, using RGB565 format.
But I can see stripes in screen as following:

Does anyone know what the problem is?
[image]

AloyseTech, 2018-12-07:
What is you background `main_color` and `grad_color`? It is possible that in RGB565 mode you only have around 16 shade of your color if the difference between the `main_color `and the `grad_color `is small. You should use dithering then and use a image background.

embeddedt, 2018-12-07:
@relliktei Are pink and orange the correct colors or are they an issue?

relliktei, 2018-12-07:
> @relliktei Are pink and orange the correct colors or are they an issue?

Pink and orange are an issue.
Contrasting to pc simulate image:
[image]

I can see triples as this:
[image]

relliktei, 2018-12-07:
> What is you background `main_color` and `grad_color`? It is possible that in RGB565 mode you only have around 16 shades of your color if the difference between the `main_color` and the `grad_color` is small. You should use dithering then and use a image background.

AloyseTech: 
Thanks for your suggesion.
I havn't use wallpager, I am using nemo theme: lv_theme_nemo_init(220, NULL) ---the same as lv_test sample code:
[image]

I will adjust the main_color and grade_color, or use wallpager to try again.

embeddedt, 2018-12-07:
@relliktei The problem must be in your display driver. Do the color depths of LittlevGL and your driver match? Possibly your driver expects BGR565 instead of RGB565?

turoksama, 2018-12-08:
It's not an issue, this is a matter of 16 color-bit.
16-bit has limited 65536 colors, when display a gradient color, we can see these gradients.
Using 24-bit should be fine.

relliktei, 2018-12-08:
> It's not an issue, this is a matter of 16 color-bit.
> 16-bit has limited 65536 colors, when display a gradient color, we can see these gradients.
> Using 24-bit or no gradient should be fine.

turoksama:

You are right, when I use 16 color depth, the pc simulator is same:
[image]

embeddedt, 2018-12-08:
@relliktei 

@turoksama is correct, the gradient issue is not really a bug.

However the colors still appear to be incorrect. Did you check what I suggested above? https://github.com/littlevgl/lvgl/issues/611#issuecomment-445261159

kisvegabor, 2018-12-08:
@relliktei I also think that the issue should be with the `RGB` `BRG` settings.
If you driver not supports the swap of color order you can use `LV_COLOR_16_SWAP   1` in lv_conf.h. It has only a minor overhead.

**About the stripe issue**
You can use a [dithered](https://blog.littlevgl.com/2017-07-03/dithering), thin vertical line as background. Dithering can be enabled in the image converter. For example:
```c
/*Gradient BG as reference*/
lv_style_scr.body.main_color = LV_COLOR_HEX(0x000048);
lv_style_scr.body.grad_color = LV_COLOR_HEX(0x004048);

/*Add the dithered image to the right*/
LV_IMG_DECLARE(bg_line_dith);                       /*10 px wide dither gradient image*/
lv_obj_t * img_dith = lv_img_create(lv_scr_act(), NULL);
lv_img_set_src(img_dith, &bg_line_dith);
lv_obj_set_width(img_dith, LV_HOR_RES / 2);         /*if object width > image source width, the image will be repeated  */
lv_obj_set_x(img_dith, LV_HOR_RES / 2);             /*Move to the rigth*/
```

Resources: [dith_bg.zip](https://github.com/littlevgl/lvgl/files/2660059/dith_bg.zip)

Result with `LV_COLOR_DEPTH  16`:
[image]

Result with `LV_COLOR_DEPTH   32` (no difference) 
[image]

relliktei, 2018-12-09:
> @relliktei I also think that the issue should be with the `RGB` `BRG` settings.
> If you driver not supports the swap of color order you can use `LV_COLOR_16_SWAP 1` in lv_conf.h. It has only a minor [...]

Dear kisvegabor:

1. I tried to set LV_COLOR_16_SWAP as 1, the result is
[image]
when I set LV_COLOR_16_SWAP as 0, the result is:
[image]

So I think the second should be what I want, this is not an RGB BGR issue.

2. Regarding with dither , I will try and report here later.
3. Before this, I try to remove gradient and set gradient color as the main color, striples are removed, but new issue appears:  color is different between pc simulaotr and real lcd device.

Then I modify source code as below again:
bg.body.main_color = LV_COLOR_HEX(0x00FF00);------->pure green
bg.body.grad_color =  LV_COLOR_HEX(0x00FF00);------->pure green
I can see there are diffent result between pc simulator and my real lcd display.

PC simulator:
[image]
This "GREEN" is pure, should be correct.

RGB LCD:

[image]
This "GREEN" is not pure, like yellow green.

Simulator lv_theme_nemo.c and embeded device lv_theme_nemo.c are same.
But the two "GREEN" are similar but SAME.
I don't know why.

embeddedt, 2018-12-09:
@relliktei Please test with pure red instead of green. In both RGB and BGR green occupies the middle 6 bits of the 16-bit value so this doesn't show whether the issue is with that.

I think the problem may actually be a combination of at least two issues.

harteware07, 2018-12-10:
have you tried to fill your buffer(s) with 0xFF on startup? Just ran into a similar issue..

embeddedt, 2018-12-10:
@harteware07 Which buffers? Can you please explain how this resolved the problem?

harteware07, 2018-12-10:
@embeddedt  I just opened a new issue explaining with bit more detail

relliktei, 2018-12-11:
> have you tried to fill your buffer(s) with 0xFF on startup? Just ran into a similar issue..

I try to init buffer with 0xff on startup, but no improve.

relliktei, 2018-12-11:
> @relliktei If the swap is not changing red to blue then something else may be wrong in your setup. Would you mind showing your display driver?

Here are my display driver code:
void lcd_fill(int32_t x1, int32_t y1, int32_t x2, int32_t y2, lv_color_t color)
{
    /*Return if the area is out the screen*/
    if(x2 < 0) return;
    if(y2 < 0) return;
    if(x1 > LCD_PIXEL_WIDTH - 1) return;
    if(y1 > LCD_PIXEL_HEIGHT - 1) return;
	
	//color.full = 0xF800;

    /*Truncate the area to the screen*/
    int32_t act_x1 = x1 < 0 ? 0 : x1;
    int32_t act_y1 = y1 < 0 ? 0 : y1;
    int32_t act_x2 = x2 > LCD_PIXEL_WIDTH - 1 ? LCD_PIXEL_WIDTH - 1 : x2;
    int32_t act_y2 = y2 > LCD_PIXEL_HEIGHT - 1 ? LCD_PIXEL_HEIGHT - 1 : y2;

	uint32_t x;
	uint32_t y;
	
	long int location = 0;

	uint16_t *fbp16 = (uint16_t*)s_psBufferLcd[0];
	/*Fill the remaining area*/
	for(y = act_y1; y <= act_y2; y++) 
	{
		for(x = act_x1; x <= act_x2; x++) 
		{
			location = (x) + (y) * LCD_PIXEL_WIDTH;
			fbp16[location] = color.full;
		}
	}
}

void lcd_map(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
	/*Return if the area is out the screen*/
	if(x2 < 0) return;
	if(y2 < 0) return;
	if(x1 > LCD_PIXEL_WIDTH - 1) return;
	if(y1 > LCD_PIXEL_HEIGHT - 1) return;
	
	//color_p->full = 0xF800;

	/*Truncate the area to the screen*/
	int32_t act_x1 = x1 < 0 ? 0 : x1;
	int32_t act_y1 = y1 < 0 ? 0 : y1;
	int32_t act_x2 = x2 > LCD_PIXEL_WIDTH - 1 ? LCD_PIXEL_WIDTH - 1 : x2;
	int32_t act_y2 = y2 > LCD_PIXEL_HEIGHT - 1 ? LCD_PIXEL_HEIGHT - 1 : y2;

	long int location = 0;
	uint16_t *fbp16 = (uint16_t*)s_psBufferLcd[0];
	
	for(int y = act_y1; y <= act_y2; y++) 
	{
		for(int x = act_x1; x <= act_x2; x++) 
		{
			location = (x) + (y) * LCD_PIXEL_WIDTH;
			fbp16[location] = color_p->full;
		}
	}
}

void lcd_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
	/*Return if the area is out the screen*/
	if(x2 < 0) return;
	if(y2 < 0) return;
	if(x1 > LCD_PIXEL_WIDTH - 1) return;
	if(y1 > LCD_PIXEL_HEIGHT - 1) return;
	
	//color_p->full = 0xF800;

	/*Truncate the area to the screen*/
	int32_t act_x1 = x1 < 0 ? 0 : x1;
	int32_t act_y1 = y1 < 0 ? 0 : y1;
	int32_t act_x2 = x2 > LCD_PIXEL_WIDTH - 1 ? LCD_PIXEL_WIDTH - 1 : x2;
	int32_t act_y2 = y2 > LCD_PIXEL_HEIGHT - 1 ? LCD_PIXEL_HEIGHT - 1 : y2;

	x1_flush = act_x1;
	y1_flush = act_y1;
	x2_flush = act_x2;
	y2_fill = act_y2;
	y_fill_act = act_y1;
	buf_to_flush = color_p;
	
	long int location = 0;
	uint16_t *fbp16 = (uint16_t*)s_psBufferLcd[0];
	
	for(int y = act_y1; y <= act_y2; y++) 
	{
		for(int x = act_x1; x <= act_x2; x++) 
		{
			location = (x) + (y) * LCD_PIXEL_WIDTH;
			fbp16[location] = color_p->full;
		}
	}
	lv_flush_ready();
}

Even I set all pixel color as 0xF800(red color) in driver, the result still shows little orange color.
[image]

embeddedt, 2018-12-11:
Below each `fbp16[location] = color_p->full;` add `color_p++`. I think that might be part of the problem.

turoksama, 2018-12-11:
please chk this line：
```
uint16_t *fbp16 = (uint16_t*)s_psBufferLcd[0];
```

I think it should look like this:

uint16_t *fbp16 = (uint16_t*)&s_psBufferLcd[0];

embeddedt, 2018-12-11:
@turoksama Good catch, but I think it should actually be:

```c
uint16_t *fbp16 = (uint16_t*)&s_psBufferLcd[0];
```

turoksama, 2018-12-11:
Well, ithink it is a copy paste missing in here...

relliktei, 2018-12-12:
@embeddedt
@turoksama

I cannot understand "color_p++", but
I try to read the buffer memory,
I make a breakpoint in lcd_flush and see the buffer memory is filled with 0xf800:

[image]

The result is this:
[image]

it's not pure red.

relliktei, 2018-12-18:
Dear All:

I have resolved this problem.
The color difference is caused by hardware.
I wired wrong cables with RGB565 signals.
ex)
MCU output signal r7,r6,r5,r4,r3 need connect to lcd input bus r7,r6,r5,r4,r3,not r5,r4,r3,r2,r1,r0.

kisvegabor, 2018-12-18:
@relliktei Great! I happy to hear that you found the root of the issue.

If you feel that all the questions are answered, please close the issue.

relliktei, 2018-12-19:
> @relliktei Great! I happy to hear that you found the root of the issue.
> 
> If you feel that all the questions are answered, please close the issue.

Thank you very very much.
I will close this issue.

closed 2018-12-19 by relliktei