lvgl/lvgl #612: Vertically center a text

nmathon, 2018-12-08:
Hi,

How can I vertically center a text in a label.
In the documentation, the following screenshot looks like if it's possible, but I wasn't able to make it. 
[image]

Thanks

turoksama, 2018-12-08:
Please be more specific.

embeddedt, 2018-12-08:
@nmathon You can *horizontally* align text:

> The label's text can be aligned to the left, right or middle with `lv_label_set_align(label, LV_LABEL_ALIGN_LEFT/RIGHT/CENTER)`

However, I don't see a way to vertically align text. Can you explain your use case for this feature?

embeddedt, 2018-12-08:
@nmathon You could create an `lv_cont` object, set the layout type to `LV_LAYOUT_CENTER`, and then create the label inside the container.

kisvegabor, 2018-12-08:
A simple `lv_obj_align` should be enough as well:

```c
lv_obj_t * titleContainer = lv_obj_create(mainContainer, NULL);   //lv_obj instead of lv_cont. (lv_obj uses less memory)
lv_obj_set_style(titleContainer, &lv_style_plain_color);
lv_obj_align(titleContainer, NULL, LV_ALIGN_IN_TOP_LEFT, 50, 10);
lv_obj_set_size(titleContainer, 260, 120);

lv_obj_t * lbTitle = lv_label_create(titleContainer, NULL);
lv_label_set_text(lbTitle, "Multiline text very very very long... And even longer... ");
lv_label_set_long_mode(lbTitle, LV_LABEL_LONG_BREAK);      //Set the long mode before width
lv_obj_set_width(lbTitle, 260);
lv_label_set_align(lbTitle, LV_LABEL_ALIGN_CENTER);
lv_obj_align(lbTitle, NULL, LV_ALIGN_CENTER, 0, 0);   //Align to the center
```