lvgl/lvgl #615: Proposition: init frame buffer

harteware07, 2018-12-10:
Hello,
I am using double buffering in comination with lv_vdb_set_adr() and just ran into an issue where the display backgrouns was full of weird colors. Filling the buffers with 0xFF eliminated the problem. Too bad it took a day to get there.. So my proposition would be to do this in lv_init() for example and make life easier.
Greetings

embeddedt, 2018-12-10:
@harteware07 Does this problem only appear when using double buffering or does it apply to a single framebuffer as well?

harteware07, 2018-12-10:
happens with both options

embeddedt, 2018-12-10:
@harteware07 I have never seen this issue. Can you show a complete example where the display background has weird colors? If this is a LittlevGL issue you should be able to reproduce it on the `pc_simulator` as well.

kisvegabor, 2018-12-10:
@harteware07 
There was an issue earlier with 32 bit color depth because the alpha channel of colors was not initialized to `0xFF` by default. However, it was about a half year ago.  Maybe have you faced with this?

harteware07, 2018-12-11:
I will update to latest version and test again!

harteware07, 2018-12-11:
I updated to 5.2 and still the same problem. But I think the problem is not linked to the missing init. It also happens when I move objects or drag them on the screen: the old buffer content is still visible, along with the new content. I attach a picture where you can see the effect. I draw a button and make it draggable. I then drag it around a little bit, you can see the image gets redrawn but the old content is not erased from the buffer. Any idea what I could try? The buffer is in external SRAM.
Thank you for your input!
[image]

embeddedt, 2018-12-11:
@harteware07 Just had a thought: are you using cache?

harteware07, 2018-12-11:
what cache do you mean?
I define the buffer with 
`static __IO uint8_t __attribute__((section (".SDRAM"))) lvgl_buff1[LV_VDB_SIZE_IN_BYTES];`
then i pass the pointer to the buffer with 
`lv_vdb_set_adr((void *)lvgl_buff1, NULL);`

The interesting thing is that it worked on the discovery board. On my custom board it fails?! Any ideas what I could try?

embeddedt, 2018-12-11:
@harteware07 Did you enable cache on the processor (i.e. is it an STM32F7, etc)?

kisvegabor, 2018-12-11:
Why are the horizontal borders are missing? 

The old content should be overdrawn by the white background. Maybe the two issue is related.

Please try this:
```c
    lv_init();

    lv_color_t c[5000];
    uint16_t i;
    for(i = 0; i < 5000; i++) c[i] = LV_COLOR_RED;

    your_disp_flush(10, 40, 60, 80, c);

    while(1);
```

harteware07, 2018-12-11:
@embeddedt  yes, it is an stm32f746, I have caching enabled. Tried to disable the i-cache, but same problem
@kisvegabor  I don't know why the h borders are missing. it is really strange. Also what happens, is when i move an item, sometimes it appears 2 times, next to each other. I will try your test code tomorrow and report back.

embeddedt, 2018-12-11:
@harteware07 The I-Cache can be left enabled. You need to disable the D-Cache. My dev board is an F746G-DISCO and the display doesn't work right unless you disable D-Cache or configure the MPU to turn off caching on the framebuffer.

harteware07, 2018-12-11:
@embeddedt does the D-cache work on the external SRAM? If yes, how could I tell the MPU not to use it on certain variables?

kisvegabor, 2018-12-12:
What happens if:
- frame buffer in SRAM
- VDB in internal RAM (e.g. with `screen_size / 20` size)
- No GPU

embeddedt, 2018-12-12:
@kisvegabor Putting the framebuffer in SRAM may not be an option. It may be too large to fit.

harteware07, 2018-12-12:
I did testing with
- frame buffer in SRAM
- VDB in internal RAM (e.g. with screen_size / 20 size)
- No GPU
The problems are the same, but plus a new error on the second (aka wrong) slider, a fine black line
[image]

harteware07, 2018-12-12:
@kisvegabor  i also tried your idea of 
`    lv_init();

    lv_color_t c[5000];
    uint16_t i;
    for(i = 0; i < 5000; i++) c[i] = LV_COLOR_RED;

    your_disp_flush(10, 40, 60, 80, c);

    while(1);`

but nothing happened at all, the display remained white. Not sure what that means?

embeddedt, 2018-12-12:
@harteware07 Sounds like the contents are not written to the display. Are you using DMA or any other type of acceleration?

harteware07, 2018-12-12:
yes i use DMA. I will roll back to the disco board hw and if this doesnt help to an old version of lvgl and see if the problem persists.

embeddedt, 2018-12-12:
@harteware07 It's not the version of LittlevGL that's the problem if @kisvegabor's test code does not function correctly. I haven't actually tried it myself but it should work.

Can you do a test without DMA?

harteware07, 2018-12-13:
I have it working now without GPU. The problem was that I changed data alignment in the DMA controller a while back because I was working on video playback. I changed it to word, lvgl was counting in half words, hence the doubling of newly rendered elements :/
So thats something stupid but still good..
The other problem, that the elements are not drawn correctly still happens when using GPU. Any ideas what could be wrong here? 
Again thanks for all your help!

embeddedt, 2018-12-13:
@harteware07 Maybe the DMA (ChromART?) and LittlevGL disagree on the color format (i.e. 16-bit vs 24-bit).

turoksama, 2018-12-13:
Hi,
Wanna try ARGB mode in gpu dma?

harteware07, 2018-12-13:
I had to switch to 32 Bit color ARGB8888 because alignment of the external SDRAM is 32 Byte and in 16Byte color mode the system crashed on redraw. GPU Mode (aka DMA2D) is still disabled due to the wrong rendering. Can anyone tell how much performance gain can be expected from GPU mode?

embeddedt, 2018-12-13:
@harteware07 I'm using F746G-DISCO without DMA and if I keep the VDB large enough I never need DMA.

harteware07, 2018-12-13:
@embeddedt my VDBs are the size of the screen so it should be fine? Btw. I dont quite understand the difference betwwen GPU mode versus "normal" DMA mode. What is the DMA2D doing differently from the DMA already running?

embeddedt, 2018-12-13:
@harteware07 I think GPU mode is capable of doing more complex graphical effects (i.e. blending) whereas standard DMA just copies bytes from place to place.

If your VDB is the same size as the screen I think performance won't be an issue.

kisvegabor, 2018-12-14:
In case of large displays, the GPU (DMA2D) can help even when filling the background with a color. LittlevGL has only a simple GPU interface (fill and blend) right now but using it resulted in a really big difference on an 800x480 display.  
If you have big areas on a wallpaper GPU blending can definitely speed up things.

But first, let's make work the most simple set-up without DMA and DMA2D.

kisvegabor, 2019-01-05:
@harteware07 Is there any news?

harteware07, 2019-01-07:
> I have it working now without GPU. 
GPU Mode is still not working but for this issue i can open a new topic when needed. So this topic cn be closed, thanks a lot to all who helped me figure it out!

kisvegabor, 2019-01-07:
Great!

closed 2019-01-07 by kisvegabor