lvgl/lvgl #617: How can I use LV_INDEV_TYPE_BUTTON?

longdadouzi, 2018-12-11:
I have four buttons, up and down and left and right. But my clock doesn't understand how to relate it to a control. I don't understand the meaning of the object group. Whenever I use the object group, it will overwrite the screen with a layer of yellow. I used the code of the day. But can't run the button correctly。

1.  /*register input Keybord*/
  indev_drv.type      =LV_INDEV_TYPE_BUTTON;
  indev_drv.read      =keyboard_read;
  lv_indev_drv_register(&indev_drv);

2.
` bool _key_pressed()
{
  u16 KEY_VALUE =~(GPIOE->IDR|0xFF86);
  if(KEY_VALUE)
  {
    return true;
  }
  return false;
}
bool keyboard_read(lv_indev_data_t *data) 
{
  data->key=Keyvalue.Keyvalue;
  if(_key_pressed())
  {
     data->state = LV_INDEV_STATE_PR;
    
  }
  else
  {
    data->state = LV_INDEV_STATE_REL;
  }
  /*No buffering so no more data read*/  
  return false;
}`
Then created the object group：
lv_group_t *Maingroup = lv_group_create();
How should it be next?

seyyah, 2018-12-11:
Issue maybe related with

- How to use lv_indev_drv_t with hardware buttons? #527
- use LV_INDEV_TYPE_BUTTON cause crash #567

longdadouzi, 2018-12-11:
I have read the contents of #527. but still can't understand

longdadouzi, 2018-12-12:
I seem to understand, is it necessary to call lv_indev_set_group (lv_indev_t *indev, lv_group_t *group);
But I don't understand how to assign the object group to the input device. Where does lv_indev_t *indev come from? Do you need to add it yourself, or have the input device specified by GL?

longdadouzi, 2018-12-12:
My problem seems to have been solved.

longdadouzi, 2018-12-12:
Can I get the key value in the button callback?

MichaelVLV, 2018-12-12:
> Can I get the key value in the button callback?

```C
static lv_res_t main_measure_action(lv_obj_t * list_btn)
{
    printf("Measure:%s\n", lv_list_get_btn_text(list_btn));

    scr.active = MEASURE_SEL_TYPE;

    return LV_RES_OK; /*Return OK because the list is not deleted*/
}

    // ---------- BUTTON 1
    lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);         /*Create a button on the currently loaded screen*/
    lv_obj_set_size(btn1, 100, 24);
    lv_obj_align(btn1, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);

    /*Create a label on the button (the 'label' variable can be reused)*/
    lv_obj_t * label = lv_label_create(btn1, NULL);
    lv_label_set_text(label, "Button1");
    lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, btn_callback); 
    lv_obj_set_free_num(btn1, 1);               /*Set a unique number for the button*/
    lv_group_add_obj(main_menu_group, btn1);
```

longdadouzi, 2018-12-13:
Hi MichaelVLV
Well, all I'm saying is get the LEFE,RIGHT.

longdadouzi, 2018-12-13:
Because I have five buttons, I want him to be a button, I can only let him press ENTERN to enter the execution I want, otherwise it will not respond

longdadouzi, 2018-12-13:
I want to get the  ‘data->key  value, can I write logic in the callback function?

embeddedt, 2018-12-13:
@longdadouzi In `lv_core/lv_indev.h`, there is a function:

```c
/**
 * Get the last key of an input device (for LV_INDEV_TYPE_KEYPAD)
 * @param indev pointer to an input device
 * @return the last pressed key (0 on error)
 */
uint32_t lv_indev_get_key(const lv_indev_t * indev);
```

Maybe it can help.

kisvegabor, 2018-12-13:
And to get the current `indev` use `lv_indev_get_act()`.

longdadouzi, 2018-12-14:
Thank you 
@MichaelVLV 
@embeddedt 
@embeddedt 
@seyyah 
My problem seems to have been solved

closed 2018-12-14 by embeddedt