lvgl/lvgl #620: `disp_flush` and data alignment

classic-gentleman, 2018-12-11:
Hi,

I'm facing a situation where the DMA engine of this MCU can only cope with 32-bit aligned transfers _and_ has weird byte packing.

Two things, then:
* Because of the byte packing oddities I have to swap the pixels (not the bytes) so that a pair of pixels are sent odd, first, even second. Attempts to away the byte packing problem with `vdb_wr` turned out to cause a _massive_ performance drop, so I'm trying to understand the moments where `lvgl` writes to the frame buffer and just do it in place. My current heuristic is `lv_draw_vbasic.c`, "everywhere `disp->driver.vdb_wr` is called", but that's not enough given I'm seeing artefacts.

* Is there a way to force the dirty areas to always produce "nicely-shaped" windows in a way I can manipulate somehow, so that when multiplied by the pixel byte count I get 32-bit aligned transfers? I'm using 16-bit colours, so I'm reaching for even-width windows (I kind of don't care about saving a column of pixels worth of data, making the DMA happy more than makes up for it).

Tips?

upbeat27, 2018-12-12:
I can point you in the right direction for your 2nd issue.
See the function `lv_refr_set_round_cb` in lv_refr.h.
This allows you to add your own callback to adjust the coordinates of the invalidated area, i.e. making the width always even. The drawback is that you might have to draw a lot more than 1 extra column of pixels, as it could end up invalidating another object and stretching the invalid area further. It's better than having DMA not work, though.

kisvegabor, 2018-12-12:
If you replace all occurrence of `vdb_wr` with your custom write what will you win? I don't think the call of `vdb_wr` function makes the massive performance loss but the implementation of pixel swapping which need to be done either in `vdb_wr` or instead of it.
I think it would be faster to swap all pixels in you `disp_flush` function. However, the `color_p` array is `const` it's safe to modify it directly.

Regarding the second question I agree with @upbeat27 , you should use `lv_refr_set_round_cb` but be aware of the mentioned issues.

closed 2018-12-12 by classic-gentleman