lvgl/lvgl #623: Crash when swithcing between screenes

MichaelVLV, 2018-12-12:
Hi.

I'm trying make switching between some screenes (lvgl run on MCU with FreeRTOS, not PC simulator).
Basic switching works good. But after some switchings (less then 10) I got crash.
Not sure where does problem is, maybe some memory leak (heap remains unchanged...)?

I have wrote simple tester.
Creating lv_task inside my freertos task:

```C
void lvgl_tft_task(void* arg)
{
	powerHold();
	LCD_power_on();
	lv_init();
	disp_spi_init();
	st7735_init();

	lv_disp_drv_t disp;
	lv_disp_drv_init(&disp);
	disp.disp_flush = st7735b_flush;
	lv_disp_drv_register(&disp);
	buttons_drv_reg;
       // switch each 1 second
	lv_task_create(scr_test_task, 1000, LV_TASK_PRIO_MID, NULL); 

	esp_register_freertos_tick_hook(lv_tick_task);

	while(1)
	{
		vTaskDelay(1);
		lv_task_handler();
	}
}
```

embeddedt, 2018-12-12:
@MichaelVLV You need to use a mutex in a multithreaded setup.

MichaelVLV, 2018-12-12:
> @MichaelVLV You need to use a mutex in a multithreaded setup.

I have added mutex as described here [https://doc.littlevgl.com/#Porting](https://doc.littlevgl.com/#Porting) in section **Using with an operating system**:

```C
void lvgl_tft_task(void* arg)
{
	powerHold();
	LCD_power_on();
	lv_init();
	disp_spi_init();
	st7735_init();

	lv_disp_drv_t disp;
	lv_disp_drv_init(&disp);
	disp.disp_flush = st7735b_flush;
	lv_disp_drv_register(&disp);
	buttons_drv_reg();

	xSemaphore = xSemaphoreCreateMutex();//xSemaphoreCreateMutex();

	if( xSemaphore == NULL )
	{
		printf("%s: mutex not created\n", __func__);
	}

       // switch each 1 second
	lv_task_create(scr_test_task, 1000, LV_TASK_PRIO_MID, NULL); 

	esp_register_freertos_tick_hook(lv_tick_task);

	while(1)
	{
		vTaskDelay(1);
		while( xSemaphoreTake( xSemaphore, portMAX_DELAY ) != pdTRUE );
		lv_task_handler();
		xSemaphoreGive(xSemaphore);
	}
}
```
But still get crash after some switches. Other tasks dont use any lv_ objects/functions.

embeddedt, 2018-12-12:
@MichaelVLV `scr_test_task` is using lv functions unless I missed something. Also, are you destroying the old screen before you load the next one? You need to replace the screen before "taking it out of context".

MichaelVLV, 2018-12-12:
> @MichaelVLV `scr_test_task` is using lv functions unless I missed something. Also, are you destroying the old screen before you load the next one? You need to replace the screen before "taking it out of context".

but it created with **_lv_task_create_** .

Should I use other mutex there aswell in this case?

MichaelVLV, 2018-12-12:
> @MichaelVLV I read that part of your code incorrectly. You are right, a mutex is not necessary with `lv_task_create`.
> 
> Try using `lv_obj_del` but load the new screen before removing the old one:
> 
> ```c
>         /* scr - previous screen object */
>         /* new_scr - new screen object */
>         lv_obj_t *new_scr = lv_obj_create(NULL, NULL);
> 	lv_scr_load(new_scr);
>         if(scr)
> 	{
> 		lv_obj_del(scr);
> 		scr = NULL ;
> 	}
> 
> 	
> ```

@embeddedt Thanks for suggestion. I have made this way, so anytime at least some screen is loaded:

```C
    static lv_obj_t * scr = NULL; 
    static lv_obj_t * new_scr= NULL;
  
   ... 

    if(scr) // screen SCR exist
    {
        // create 2nd screen and load
        new_scr = lv_obj_create(NULL, NULL); 
        lv_scr_load(new_scr);
        // delete existing screen
    	lv_obj_del(scr);
        scr = NULL ;
    }
    else // no screen SCR exist or it was deleted
    {
        scr = lv_obj_create(NULL, NULL);
        lv_scr_load(scr);

        // delete second screen
    	if(new_scr)
    	{
    		lv_obj_del(new_scr);
    		new_scr = NULL ;
    	}
    }
```

But still I got crash.

MichaelVLV, 2018-12-12:
> @MichaelVLV Can you show a backtrace of where it crashes?

@embeddedt 

Here is backtrace:

```
lv_obj_create (parent=0x3ffcf808 <work_mem+32628>, copy=0x0)
    at C:/Projects/screen_project/lvgl/lv_core/lv_obj.c:173
173             new_obj->par = parent; /*Set the parent*/
```

embeddedt, 2018-12-12:
@MichaelVLV Try increasing the heap size.

MichaelVLV, 2018-12-12:
@embeddedt I have increased heap size twice, now it is 20kB already:

```C
xTaskCreate(lvgl_tft_task, "lvgl_tft_task", 1024 * 20, NULL, 10, NULL);
```
But crash happens about at same time (not longer or shorter)

embeddedt, 2018-12-12:
@MichaelVLV I meant in `lv_conf.h`.

MichaelVLV, 2018-12-12:
@embeddedt thanks for pointing. Problem was nearby :)

https://github.com/littlevgl/lvgl/blob/master/lv_core/lv_group.c#L45

this function allocates memory each time when I called it in switch statement.

I moved group creation

```C
   scr1_group = lv_group_create();
   sc2_group = lv_group_create();
```
to task initialisation.

Looks like issue is solved now.

embeddedt, 2018-12-12:
@MichaelVLV Glad to hear it! I suspected the heap was an issue when you mentioned that it crashes after several screen switches.

closed 2018-12-12 by embeddedt

kisvegabor, 2018-12-12:
I'm a little bit late but in the future enabling `USE_LV_LOG` in `lv_conf.h` can help.

MichaelVLV, 2018-12-16:
An other problem occures.
Problem is in repeatedly adding members to group I think:

```C
lv_group_add_obj(scr1_group, btn4);
```

So my logic was next.
1) create needed  groups once
```C
scr1_group = lv_group_create();
```
2) set "active" group
```C
lv_indev_set_group(emulated_kp_indev, scr1_group);
```
3) make screen (including add objects to group)
```C
	scr_prepare("scr1_menu"); // PREPARE SCREEN

	lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
	lv_obj_set_size(btn1, 100, 24);
	lv_obj_align(btn1, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);
	lv_obj_t * label = lv_label_create(btn1, NULL);
	lv_label_set_text(label, "Button1");
	lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, my_action);
	lv_group_add_obj(scr1_group, btn1);
   . . . 
```
And I got crash after loading new screen and try navigating UP/DOWN. Just before crash, on screen I dont see "selected" objects.

So I tried use this function **(added it in header _lv_core/lv_group.h_, not sure why it was absent, its not static function)**:
[https://github.com/littlevgl/lvgl/blob/master/lv_core/lv_group.c#L65](https://github.com/littlevgl/lvgl/blob/master/lv_core/lv_group.c#L65) 
and changed logic to next:
1) delete all groups 
2) create one needed group again
3) set this group as "active" 
4) load my screen (which also fills group with new objects)

@embeddedt @kisvegabor 
This solution worked for me, bit I'm not sure if this is correct way, can you give advice?

kisvegabor, 2018-12-16:
@MichaelVLV Please send a code example to reproduce the issue. Just a few lines of codes with only the required things.