lvgl/lvgl #624: Text area cursor is drawn incorrectly when LV_TXT_UTF8 is 0

upbeat27, 2018-12-12:
Here is an example of how the cursor looks with `#define LV_TXT_UTF8 1`.
The size of the cursor appears to match the letter size:
[image]
[image]

Here is what it looks like with `#define LV_TXT_UTF8 0`, with no other changes.
Things do not line up, and you can't tell what letter you are actually on:
[image]
[image]

This affects the block and outline cursor types as well (it looks even worse than these examples with the underline cursor).
The line cursor works, interestingly enough.

Using release 5.2.0.
Font is unscii.

kisvegabor, 2018-12-12:
Hi,

I fixed it in `dev-5.3`. Pease test it!

upbeat27, 2018-12-14:
Tested with `dev-5.3`, can confirm it works correctly now.

closed 2018-12-14 by upbeat27