lvgl/lvgl #625: Drag Action Callback on Obj

fishermr, 2018-12-12:
I have an img  obj with drag set to true.  No problems dragging the obj.  I have no idea how to set an action callback so I can get the new position of the obj while it's being dragged and know when it's been released.  .  Been through the doc, code, examples and google.  Any suggestions.

embeddedt, 2018-12-12:
@fishermr You could try replacing the signal function.

fishermr, 2018-12-12:
Did look into using a signal function.  This seems to be used for interrupts.  How would I use this for an event callback?

embeddedt, 2018-12-12:
@fishermr Signals are internal event callbacks: they are called by core LittlevGL when something happens on the object. I think the relevant signals for your use case would be `LV_SIGNAL_DRAG_BEGIN` and `LV_SIGNAL_DRAG_END`.

mfisher-azafran, 2018-12-13:
Thanks for the concise response.  Much appreciated.

embeddedt, 2018-12-13:
@mfisher-azafran @fishermr Were you able to make it work?

fishermr, 2018-12-13:
Working on understanding how to create the code using signals.

As seen below, I need to allow the user to drag a food object to the center for selection. While dragging, it needs to maintain its vertical position and exceed the horizontal edges.  Very simple process for obj movement.

kisvegabor, 2018-12-13:
@fishermr 
I've created a blog post about internal mechanisms of LittlevGL: https://blog.littlevgl.com/2018-12-13/extend-lvgl-objects

You can learn how to use:
- the object's extended data 
- design functions
- **signal functions**

I hope it will help!

mfisher-azafran, 2018-12-13:
Thank u

RinkuJ, 2020-12-22:
can we drag a page in inverse vertical direction?
for example: if I pull the page up then seeing the content of page above it and if i drag the page down will show the content below.

kisvegabor, 2020-12-27:
Answered here: https://github.com/lvgl/lvgl/issues/1974#issuecomment-750379999