lvgl/lvgl #628: how to set tabview's label height

3038922, 2018-12-14:
how to set tabview's label height?
The automatic label is too high. I want to change it to a smaller size.

kisvegabor, 2018-12-14:
You can set it int the `LV_TABVIEW_STYLE_BTN_REL`, `LV_TABVIEW_STYLE_BTN_PR` etc styles.

```c
style.text.font = &my_small_font;
lv_tabview_set_style(tabview, LV_TABVIEW_STYLE_BTN_REL, &style);
```

kisvegabor, 2018-12-17:
Here is a full example:
```c

    static lv_style_t style_tv_btn_rel;
    lv_style_copy(&style_tv_btn_rel, &lv_style_btn_rel);
    style_tv_btn_rel.text.font = &lv_font_dejavu_10;
    style_tv_btn_rel.body.padding.ver = 5;                  /*The vertical padding of the released button is applied to all buttons*/

    lv_obj_t * tv = lv_tabview_create(lv_scr_act(), NULL);
    lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_REL, &style_tv_btn_rel);
    lv_tabview_add_tab(tv, "First");
    lv_tabview_add_tab(tv, "Second");
```

[image]

3038922, 2018-12-17:
> Here is a full example:
> 
> ```c
>     static lv_style_t style_tv_btn_rel;
>     lv_style_copy(&style_tv_btn_rel, &lv_style_btn_rel);
>     style_tv_btn_rel.text.font = &lv_font_dejavu_10;
>     /*The vertical padding of the released button is applied to all buttons*/
>     style_tv_btn_rel.body.padding.ver = 5;                
> 
>     lv_obj_t * tv = lv_tabview_create(lv_scr_act(), NULL);
>     lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_REL, &style_tv_btn_rel);
>     lv_tabview_add_tab(tv, "First");
>     lv_tabview_add_tab(tv, "Second");
> ```
> 
> [image]

thank you .I tested it. It worked.

kisvegabor, 2018-12-18:
Great!

closed 2018-12-18 by kisvegabor

zhangshengping, 2019-03-08:
if I do not want the label, How can I do it?

kisvegabor, 2019-03-08:
You can use `void lv_tabview_set_btns_hidden(lv_obj_t *tabview, bool en)`.

---

We just started a [Forum](https://forum.littlevgl.com/) for LittlevGL. Please use it to ask questions.
The GitHub issues should be about development related things.

See: https://github.com/littlevgl/lvgl/issues/906