lvgl/lvgl #629: Group input not updating slider

BrianPugh, 2018-12-14:
All the proper callbacks are triggering, but the `lv_slider` value isn't updating. It used to work; investigating.

kisvegabor, 2018-12-14:
Hi,

Thank you for the report. I found the root of the issue. In the near past `indev_keypad_proc` was updated to send `LV_SIGNAL_PRESSED` on `LV_GORUP_KEY_ENTER` to make possible to use of the button's press action. However, it broke the slider.

I fixed it in https://github.com/littlevgl/lvgl/commit/6f1d0ef97b9ee993d16420d94585ba5a1b195bf6

Does it work there too?

kisvegabor, 2018-12-14:
Have you updated to latest version with my fix?

BrianPugh, 2018-12-14:
yes, commit 6f1d0ef97 seems to have fixed the slider updating issue, but it introduced a new issue where i cannot set range or values.

embeddedt, 2018-12-14:
@BrianPugh What is the value of the `lv_obj_t *` argument inside `lv_obj_get_ext_attr`?

BrianPugh, 2018-12-14:
I'll be back with more info/test-case this evening.

kisvegabor, 2018-12-14:
I tested it with `lv_test_group_1` in PC simulator and it worked. Please test it with a very minimal example with only one slider.

BrianPugh, 2018-12-15:
naively, it seems like somewhere in my code a `NULL` pointer got introduced; I'm sure this is fixed, but I'll try and double confirm it soon.

kisvegabor, 2018-12-16:
Great! Thanks for the feedback! :)