lvgl/lvgl #632: Use animation to show and hide objects

InfiniteYuan1, 2018-12-14:
Thank you in advance for your work.
I want to use animation to show and hide object, but there are some interesting things that:
1. It's ok for use animation to show object in first time.
2. It's ok for use animation to hide object when i click the object.
3. Interesting. I want to animation to show object after the animation of hide is worked, but use animation to show object is error again, the object's position on the top of screen is error.

my lvgl commit id is: b6b77f4b456e832504cf7ac364f4c456a5eb5bb3

I hope you can help me resolve the problem.

Thank you in advance.
```
………
static void play_callback(lv_obj_t * obj)
{
    lv_obj_animate(playbtn[tab_id], LV_ANIM_FLOAT_TOP | LV_ANIM_IN, 400, 0, NULL);
}

static lv_res_t playbtn_action(lv_obj_t * btn)
{
    lv_obj_animate(playbtn[tab_id], LV_ANIM_FLOAT_TOP | LV_ANIM_OUT, 400, 0, play_callback);
    return LV_RES_OK;
}

static void create_obj()
{
    ……
    playbtn[id] = lv_btn_create(parent, NULL);
    lv_obj_set_size(playbtn[id], 70, 70);
    lv_obj_t *playimg = lv_img_create(playbtn[id], NULL);
    lv_img_set_src(playimg, btn_img[1]);
    lv_obj_align(playbtn[id], parent, LV_ALIGN_IN_TOP_MID, 0, 170);
    lv_btn_set_action(playbtn[id], LV_BTN_ACTION_CLICK, playbtn_action);
    lv_obj_animate(playbtn[id], LV_ANIM_FLOAT_BOTTOM | LV_ANIM_IN, 400, 0, NULL);
    ……
}
```

embeddedt, 2018-12-14:
@InfiniteYuan1 When you use the show animation it uses the current position of the object on screen. Set the position again with `lv_obj_set_pos` and then immediately call `lv_obj_animate`.

InfiniteYuan1, 2018-12-15:
Thank you , try it.

InfiniteYuan1, 2018-12-15:
Can I use `lv_obj_align` functiong befter call `lv_obj_animate`?

embeddedt, 2018-12-15:
@InfiniteYuan1 Yes, that should work.

InfiniteYuan1, 2018-12-17:
@embeddedt Thank you , work fine.

closed 2018-12-17 by InfiniteYuan1