lvgl/lvgl #635: Feature Request: Free "free_ptr" on deletion

BrianPugh, 2018-12-16:
I think it would be useful if there was an option to call `lv_mem_free()` on the `free_ptr` object attribute upon object deletion. It seems like in most use cases, if `free_ptr` is used, it will be pointing at a dynamically created object. Perhaps this could be a runtime (or compile time) configuration. I'm willing to take a stab at making a PR, but wanted to refine the idea first.

embeddedt, 2018-12-16:
@BrianPugh What about the case where `free_ptr` points to a dynamically created object not on the LittlevGL heap?

BrianPugh, 2018-12-16:
What if there's an attribute for a free-like function pointer. If it's not NULL, call it upon deletion with `free_ptr` as the sole parameter. If it's NULL, perform no cleanup action on the `free_ptr`

embeddedt, 2018-12-16:
@BrianPugh That sounds better, but why couldn't the user free the object before calling `lv_obj_del`?

BrianPugh, 2018-12-16:
This will handle for the case where the object's parent is deleted. Otherwise the programmer will have to recursively search the children and check their `free_ptr` attribute before deleting.

embeddedt, 2018-12-16:
@BrianPugh Good point; didn't think of that. I think you should go ahead and make the PR, since it shouldn't be all that much work and it can always be revised before merging.

Maybe the function pointer type should be called `lv_free_t` or something along those lines.

BrianPugh, 2018-12-16:
So with this, the namespace is actually getting a bit confusing. Do you have more recommendations on names?

The free function handle type could be `lv_free_ptr_free_t` and the attribute could be named `free_ptr_free`. I sort of wish `free_ptr` and `free_num` were named something along the lines of `user_ptr` and `user_num` to avoid this overloading.

embeddedt, 2018-12-16:
@BrianPugh For the record, `free_ptr` and `free_num` will be removed sometime in the future and replaced by a new system where you specify exactly what kind of data you want to store.

For now, maybe you could call it a deleter? (Thinking along the lines of C++)

BrianPugh, 2018-12-16:
is there any conversation/documentation/code on the future system? I just found out about `free_ptr` and `free_num` and it makes a lot of my code much cleaner. Will the new system be relatively similar?

I think `lv_free_ptr_deleter_t` is a pretty good name.

embeddedt, 2018-12-16:
@BrianPugh 

> Is there any conversation/documentation/code on the future system?

Read from here down: https://github.com/littlevgl/lvgl/issues/316#issuecomment-444404195

> I think `lv_free_ptr_deleter_t` is a pretty good name.

Great!

kisvegabor, 2018-12-16:
The solution should be compatible with the new `free_data` vision. 

Storing a new function to free the `free_data` would need an extra 4 bytes/object but usually, this feature is not required. 

What about extending the signal function's `LV_SIGNAL_CLEAN_UP` signal?

embeddedt, 2018-12-16:
@kisvegabor I think that would not require any changes to LittlevGL to implement and it would still work with `free_data`. Good idea!

BrianPugh, 2018-12-16:
Extending the signal function seems like the most universal, powerful solution. Thanks guys!

closed 2018-12-16 by BrianPugh

kisvegabor, 2018-12-16:
Happy to hear that you can use this solution.
I've written a guide about signal functions and similar things: https://blog.littlevgl.com/2018-12-13/extend-lvgl-objects  Maybe it'll help.