lvgl/lvgl #637: Bug: lv_label child of lv_bar will not draw

BrianPugh, 2018-12-16:
Creating a `lv_label` with a parent of `lv_bar` causes the label to not be displayed.

kisvegabor, 2018-12-16:
It works for me:
```c
    lv_obj_t * bar = lv_bar_create(lv_scr_act(), NULL);
    lv_bar_set_value(bar, 30);
    lv_obj_align(bar, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_obj_t * label = lv_label_create(bar, NULL);
    lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);
```

[image]

Maybe the label's color is the same as the background color of the bar?

BrianPugh, 2018-12-16:
I think my issue is the inhereted style of the label from the bar. I haven't solved this yet, but I think its my problem rather than an lvgl problem. Thanks!

closed 2018-12-16 by BrianPugh