lvgl/lvgl #638: eInk optimizations

embeddedt, 2018-12-16:
Continued from #109:

> What I meant was whether the design of LittlevGL requires a certain refresh rate in order for things to work properly (i.e. showing button toggles).

> I ask because most eInk screens cannot handle high refresh rates without some hacking, which I haven't gotten into yet. Therefore, I lowered LV_REFR_PERIOD in lv_conf.h. I then noticed that while button actions were still called, I could no longer see the visual toggle effect.

> Is there a way to force a refresh even outside of LV_REFR_PERIOD?

kisvegabor, 2018-12-16:
> Therefore, I lowered LV_REFR_PERIOD in lv_conf.h. I then noticed that while button actions were still called, I could no longer see the visual toggle effect.

It was not visible because you didn't press the button longer than `LV_REFR_PERIOD`?

As I see these are contrary things:
1. Lower refresh rate
2. Have a refresh rate fast enough for quick button presses.

Anyway, maybe `lv_refr_now()` might help. It will call the refresh task to make a refresh immediately.

embeddedt, 2018-12-16:
@kisvegabor Can `lv_refr_now` be called in the `indev_read` context?

I really want to call it immediately after the touch is detected and processed by LittlevGL. How would you recommend doing that?

kisvegabor, 2018-12-16:
You can call in the action. 

There was a [discussion](https://github.com/littlevgl/lvgl/issues/435) about a "feedback" callback. It would be triggered when a user interacts with the object and in the feedback callback he can play a "click" sound or start a vibro motor.  

It seems easy to implement and it can be used the react to the user action with `lv_refr_now()`

kisvegabor, 2018-12-16:
It would be great! 

My view was:
```c
lv_indev_set_feedback(indev, my_feedback);

void my_feedback(lv_indev_t * indev, lv_signal_t signal)
{
   if(signal == LV_SIGNAL_RELEASED) {
   /*Do something*/
   }
}
```

I have an idea how to easily  implement maybe it helps. In `lv_obj_signal()` is `lv_indev_get_act() != NULL` call the feedback with the related signals.

Once we will introduce the new actions system I think `lv_action_type_t` would be better instead of `lv_signal_t`

embeddedt, 2018-12-20:
@kisvegabor Yes, thank you for merging it!

closed 2018-12-20 by embeddedt

kisvegabor, 2018-12-20:
As I see, working with eInk displays seems challenging. Would you e interested in writing a blog post about your experiences?

embeddedt, 2018-12-20:
@kisvegabor I would be happy to do that. How do I create a post?