lvgl/lvgl #639: Probot

embeddedt, 2018-12-16:
This is just a suggestion, feel free to not implement it.

Currently we do a lot of manual organization of issues. Could [Probot](https://probot.github.io/) be useful for this?

kisvegabor, 2018-12-17:
Hm, looks useful. Especially the `stale` extension: https://probot.github.io/apps/stale/

It seems it can be customized very well. We can use `architecture` label for "never consider stale" 

```yml
# Number of days of inactivity before an issue becomes stale
daysUntilStale: 60
# Number of days of inactivity before a stale issue is closed
daysUntilClose: 7
# Issues with these labels will never be considered stale
exemptLabels:
  - pinned
  - security
# Label to use when marking an issue as stale
staleLabel: wontfix
# Comment to post when marking an issue as stale. Set to `false` to disable
markComment: >
  This issue has been automatically marked as stale because it has not had
  recent activity. It will be closed if no further activity occurs. Thank you
  for your contributions.
# Comment to post when closing a stale issue. Set to `false` to disable
closeComment: false
```

kisvegabor, 2019-01-05:
I've added Stale. Let's see how it works!
See: https://github.com/littlevgl/lvgl/blob/master/.github/stale.yml

closed 2019-01-05 by kisvegabor

embeddedt, 2019-01-05:
@kisvegabor Sounds good!