lvgl/lvgl #640: Performance issue with large amounts of data in text area

embeddedt, 2018-12-16:
When the text area contains large amounts of data, the whole UI becomes very slow.

To reproduce this, you can create a 16 kilobyte array filled with random characters and pass it to `lv_ta_set_text`.

Is the text area designed for large text files like this?

kisvegabor, 2018-12-17:
I added a fix to `dev-5.3` to speed up the drawing of long texts. 
Does it help?

embeddedt, 2018-12-17:
@kisvegabor I didn't use the benchmark, so I can't give exact statistics. It seems a bit faster, but overall the experience would still be bad for a user.

Did you test with code similar to the following?

```c
        static char random_data[16384];
	for(int i = 0; i < sizeof(random_data) - 1; i++)
		random_data[i] = 'A' + (rand() % ('Z' - 'A'));
	lv_ta_set_text(ta, random_data);
```

kisvegabor, 2018-12-17:
I did it like this (quite the same):
```c
 char buf[16 *1024 + 1];
    char txt[]  = {"abcd"};
    buf[16*1024] = '\0';

    uint32_t i;
    for (i = 0; i < 16 * 1024; i++) {
        buf[i] = txt[i%4];
    }

    lv_obj_t * ta = lv_ta_create(lv_scr_act(), NULL);
    lv_ta_set_text(ta, buf);
    lv_obj_set_pos(ta, 20, 20);
```

However, you didn't close the string! Maybe a lot of unprintable characters are there until a  `\0` comes.

embeddedt, 2018-12-17:
@kisvegabor Thanks for catching that. I wrote the random data code quickly during the debugging process so I didn't think of the null byte. I will fix it and get back to you with the result.

embeddedt, 2018-12-18:
@kisvegabor Unfortunately there is no improvement. I am still generating roughly 16 kilobytes of text and the text area is still slow. It may not be a rendering problem as the text area is only 90 pixels tall.

embeddedt, 2018-12-20:
@kisvegabor How would I immediately return from `lv_draw_label`? Should I put a hack in `lv_ta`? I am not calling `lv_draw_label` directly.

kisvegabor, 2018-12-20:
I mean hack `lv_draw_label` in `lv_draw_label.c`:

```c
 */
void lv_draw_label(const lv_area_t * coords, const lv_area_t * mask, const lv_style_t * style, lv_opa_t opa_scale,
                   const char * txt, lv_txt_flag_t flag, lv_point_t * offset)
{
  return;   /*Prevent the drawing of letters*/

  ....
}
```

embeddedt, 2018-12-20:
@kisvegabor Performance significantly improved with that hack. Although the UI is still slower than normal, it is definitely usable now.

embeddedt, 2018-12-21:
@kisvegabor Should have thought of the cursor! I will do some more experimenting and get back to you with the results.

embeddedt, 2019-01-05:
@kisvegabor I tried it again with my test code.

1. The UI is now fast even when the text area is filled with text, so in general the issue has been resolved.
2. While removing/adding a character the UI feels sluggish.

At this point I think optimizing the text area further might be too complex.

kisvegabor, 2019-01-05:
Sluggish means a little bit slow or unusable? :)

embeddedt, 2019-01-05:
@kisvegabor A little bit slow. :wink:

kisvegabor, 2019-01-05:
I've checked the code what happens when a character is added but didn't find any obvious bottleneck.
So if it's okay, then I vote for leave it for now. :)

embeddedt, 2019-01-05:
@kisvegabor Sounds good. Thanks for the help!

closed 2019-01-05 by embeddedt

rrroonn, 2019-02-06:
Hi guys, 
I was having a similar problem, found this discussion and subsequently refreshed by lvgl local copy, from "master". Now have:
```
#define LVGL_VERSION_MAJOR   5
#define LVGL_VERSION_MINOR   3
#define LVGL_VERSION_PATCH   0
```
Problem remains.

I have traced code and see that underlying label is still invalidated for each char added - so I assume that I dont have the correct version.

Should I use the 6.0 dev library?

Some more background.

Using STM32F407VET6 with SSD1962 800x480 display. 

```
    mainTextArea = lv_ta_create(parent, NULL);
    lv_obj_set_size(mainTextArea, lv_obj_get_width(parent)-2*MAIN_PADDING_X, 
    lv_obj_get_height(parent)-MAIN_BUTTON_HEIGHT-3*MAIN_PADDING_Y);
    lv_obj_align(mainTextArea, parent, LV_ALIGN_IN_TOP_MID, 0,0);
    lv_ta_set_style(mainTextArea,LV_TA_STYLE_SB, &style_sb);                     /*Apply the scroll bar style*/
    lv_ta_set_style(mainTextArea, LV_TA_STYLE_BG, &lv_style_transp_fit);
    lv_ta_set_cursor_type(mainTextArea, LV_CURSOR_NONE);
    lv_ta_set_max_length(mainTextArea,MAX_TEXT_LENGTH);
    lv_ta_set_text(mainTextArea,"");
```
Because of memory constrains, I limited the size of the text area (currently 1800 bytes).

To deal with this, I also delete old text if needed to make room

```
// =======================================================================
/*
 * printf to the message area
 */
void log_message(const char *format,...)
{
	if (!mainTextArea) return; // gui not initialised yet

	va_list	args;
	char * buffer = malloc(80);
	 va_start(args,format);
	 vsprintf(buffer,format,args);
	 va_end(args);

	 const char * p = lv_ta_get_text(mainTextArea);
	 int need = strlen(buffer);
	 int max = lv_ta_get_max_length(mainTextArea);
	 int len = strlen(p);
	 int del = need-(max-len);

	 if (del>0)
	 { // make room - delete earliest
		 p+=del;
		 while ((*p!='\n')&&(del<max)) { del++; p++;} // delete to a line ending
		 if (*p=='\n') { del++; p++;}
		 char * temp = (char*) malloc(max-del+1);
		 if (temp)
		 {
			 memcpy(temp,p,max-del); // copy existing
			 lv_ta_set_text(mainTextArea,temp);
			 lv_ta_set_cursor_pos(mainTextArea, LV_TA_CURSOR_LAST);
			 free(temp);
		 }
	 }

	 lv_ta_add_text(mainTextArea, buffer);

	 free(buffer);
}

```
The performance issues don't seem related to deleting the chars, though it is obviously inefficient; however, I dont see another way to do it.  I fixed buffer could be a solution?  Anyway, performance is an issue regardless. When running on the MCU, I can wait 6 seconds for a line to be logged.  A lot of cpu cycles going somewhere, but I have a breakpoint on the SSD1963 flushing code and it is not being called during the process of updating the text.  Unfortunately, I dont have any way of profiling under windows or atollic.

Maybe I am missing something obvious, or not using something correctly?

Appreciate any thoughts and ideas?
thank you
Ron

kisvegabor, 2019-02-07:
1. How many characters do you have in the text area when you experience the 6 seconds delay?
2. What happens if you just add a lot of lines in a `for` cycle without any extra logic?

rrroonn, 2019-02-07:
1. 1800 chars
2. still long delays - many seconds

test code:
```
	if (evt.status == osEventMessage)
	{
		T_MSG * m = (T_MSG *)evt.value.p;
		log_message("RX: %s\n",m->msg);
		osPoolFree(pool,m);
	}
	else if ((loops++%1000)==0)
	{
		for (int i=0; i<30; i++)
		{
			log_message("Loop %5d((%d)================================================\n",loops/1000,i);
		}

	}

```
```
void log_message(const char *format,...)
{
	if (!mainTextArea) return; // gui not initialised yet

	va_list	args;
	char * buffer = malloc(80);

	 va_start(args,format);
	 vsprintf(buffer,format,args);
	 va_end(args);

	 const char * p = lv_ta_get_text(mainTextArea);
	 int need = strlen(buffer);
	 int max = lv_ta_get_max_length(mainTextArea);
	 int len = strlen(p);
	 int del = need-(max-len);

	 if (del>0)
	 { // make room - delete earliest
		 p+=del;
		 while ((*p!='\n')&&(del<max)) { del++; p++;} // delete to a line ending
		 if (*p=='\n') { del++; p++;}
		 char * temp = (char*) malloc(max-del+1);
		 if (temp)
		 {
			 memcpy(temp,p,max-del); // copy existing
			 lv_ta_set_text(mainTextArea,temp);
			 lv_ta_set_cursor_pos(mainTextArea, LV_TA_CURSOR_LAST);
			 free(temp);
		 }
/* previous method - just delete chars
		 lv_ta_set_cursor_pos(mainTextArea,del);
		 for (int i=0; i<del; i++)
		 {
			 lv_ta_del_char(mainTextArea);
		 }
		 p = lv_ta_get_text(mainTextArea);
		 lv_ta_set_cursor_pos(mainTextArea,strlen(p));
*/
	 }

	 lv_ta_add_text(mainTextArea, buffer);

	 free(buffer);
}

```

So I wrote some new test code:
```
    uint32_t total=0;
	uint32_t delta=0;
	for (int i=0; i<50; i++)
	{
		start_us();
		just_add_message("%3d: Elapsed: %4ldmSec, last: %4ldmSec ==========================\n",i,total,delta);
		delta = elapsed_us()/1000;
		total += delta;
	}

```
```
void just_add_message(const char *format,...)
{
	if (!mainTextArea) return; // gui not initialised yet

	va_list	args;
	char * buffer = malloc(80);

	 va_start(args,format);
	 vsprintf(buffer,format,args);
	 va_end(args);

	 lv_ta_add_text(mainTextArea, buffer);

	 free(buffer);
}
```

This loop of 50 iterations took  69.5 seconds with the following profile:
```
Loops:
49: 2787 mSec
48: 2730
40: 2275
37: 2104
---
10: 506
01: 68
etc
```
note that I increased the text buffer to 

```
#define MAX_TEXT_LENGTH     (60*80)
lv_ta_set_max_length(mainTextArea,MAX_TEXT_LENGTH);

```
I have done a bit of tracing and noted that _each added char_ gets processed substantially, including memory being reallocated, size being readjusted etc.

It would be nice to have a simplified raw text area, fixed visual size and using an external buffer with a refresh()/redraw() method.  Having text wrapping and scrollbars would be ideal for me.

I can see the purpose of the Text Area is primarily to allow text entry so I understand that my use is not really the target use-case.

kisvegabor, 2019-02-10:
> I have done a bit of tracing and noted that _each added char_ gets processed substantially, including memory being reallocated, size being readjusted etc.

It's so slow because `lv_ta_set_max_length` is used and characters are really checked one by one. If you turn it off (set the max length to 0) `lv_ta_add_text` should be much faster. Can you test it?

rrroonn, 2019-02-10:
Ok, I have done the requested additional testing and it is much faster (as expected).

Line 49: 50mSec
Line 30: 30mSec
Line 20: 20 mSec
etc

However, I changed to the fixed size TA to get away from memory exhaustion problems, as each time you add text to the TA, it creates a copy of its contents so it can reallocate memory.  If its 4k of text, then you need 8k memory to support this.  Not really a problem for small text areas, but did cause me grief.

kisvegabor, 2019-02-12:
Yes, really there is a point when the previous and the current memory exists in `lv_mem_realloc`.

What you can do is a little trick:
1. Set the text to a lot of `space` characters
2. When you need to update the text area get its text like this:
```c
    lv_obj_t * label = lv_ta_get_label(ta);
    char * txt = (char *)lv_label_get_text(label);    /*Drop const */
```
3. Add your text to `txt`
4. Refresh the label: `lv_label_set_text(label, NULL);`

I didn't test it, but something similar should work.