lvgl/lvgl #642: Adding duplicate buttons when adding buttons to window components.

chaos4284, 2018-12-17:
void component_test()
    {    
       static lv_style_t style_sb;
       static lv_obj_t *test;
       static lv_obj_t *my_setup_action;
       static lv_obj_t *lv_win_close_action;
       lv_style_copy(&style_sb, &lv_style_plain);
       style_sb.body.main_color = LV_COLOR_BLACK;
       style_sb.body.grad_color = LV_COLOR_BLACK;
       style_sb.body.border.color = LV_COLOR_WHITE;
       style_sb.body.border.width = 1;
       style_sb.body.border.opa = LV_OPA_70;
       style_sb.body.radius = LV_RADIUS_CIRCLE;
       style_sb.body.opa = LV_OPA_60;
    
       /*Create a window*/
       lv_obj_t * win = lv_win_create(lv_scr_act(), NULL);
       lv_win_set_title(win, "Example window"); /*Set the title*/
       lv_win_set_style(win, LV_WIN_STYLE_SB, &style_sb); /*Set the scroll bar style*/
       /*Add control button to the header*/
       my_setup_action = lv_btn_create(win,NULL);

       test = lv_btn_create(win,NULL);    
    
       lv_win_add_btn(win, SYMBOL_SETTINGS, my_setup_action); /*Add a setup button*/
       lv_obj_align(test, NULL, LV_ALIGN_CENTER, 0,0);    

       lv_obj_t * txt1 = lv_label_create(win, NULL);
       lv_label_set_text(txt1, "This is the content of the window\n\n"
       "You can add control buttons to\nthe window header\n\n"
       "You can scroll it\n\n"
       "See the scroll bar on the right!");
    } 
[image]

kisvegabor, 2018-12-17:
I don't see clearly what was your intention with `my_setup_action` but should be a function like:
```c
static lv_res_t my_setup_action(lv_obj_t * win_btn)
{
    lv_obj_t * win = lv_win_get_from_btn(win_btn);
    lv_obj_t * new_btn = lv_btn_create(win, NULL);

    return LV_RES_OK;
}
```

chaos4284, 2018-12-18:
My intention is to place a window setting button via SYMBOL_SETTINGS and a button at a specific position in the window.

Therefore, I used lv_obj_align () function to specify a specific position in a window with one button.

After checking the GUI, I confirmed that one more unintended button was added.

In Windows, I would like to ask you if there is another function to place a button at a specific location.
[image]

kisvegabor, 2018-12-18:
Here is a complete example based on your code:
```c
static lv_res_t my_setup_action(lv_obj_t * win_btn)
{
    lv_obj_t * win = lv_win_get_from_btn(win_btn);
    lv_obj_t * new_btn = lv_btn_create(win, NULL);
///    Align to the center
//    lv_obj_align(new_btn, NULL, LV_ALIGN_CENTER, 0,0);

//  Or align to the top middle
    lv_obj_align(new_btn, NULL, LV_ALIGN_IN_TOP_MID, 0, 20);
 
   return LV_RES_OK;
}

void component_test(void)
{
   static lv_style_t style_sb;
   lv_style_copy(&style_sb, &lv_style_plain);
   style_sb.body.main_color = LV_COLOR_BLACK;
   style_sb.body.grad_color = LV_COLOR_BLACK;
   style_sb.body.border.color = LV_COLOR_WHITE;
   style_sb.body.border.width = 1;
   style_sb.body.border.opa = LV_OPA_70;
   style_sb.body.radius = LV_RADIUS_CIRCLE;
   style_sb.body.opa = LV_OPA_60;

   /*Create a window*/
   lv_obj_t * win = lv_win_create(lv_scr_act(), NULL);
   lv_win_set_title(win, "Example window"); /*Set the title*/
   lv_win_set_style(win, LV_WIN_STYLE_SB, &style_sb); /*Set the scroll bar style*/

   /*Add control button to the header*/
   lv_win_add_btn(win, SYMBOL_SETTINGS, my_setup_action); /*Add a setup button*/

   lv_obj_t * txt1 = lv_label_create(win, NULL);
   lv_label_set_text(txt1, "This is the content of the window\n\n"
   "You can add control buttons to\nthe window header\n\n"
   "You can scroll it\n\n"
   "See the scroll bar on the right!");
}
```

[image]

chaos4284, 2018-12-18:
The code above is an example of creating a button when you click the Settings button.

The situations I want are screen1 and screen2.
At this time, when I click the button of screen1, I want a situation where the setting button and one additional button are simultaneously displayed on the window in screen2.
That is, as follows:
[image]

kisvegabor, 2018-12-19:
I see. It should be quite simple. Just create a button on `scr1` set its action and add the window button and create the normal in the action:
```c

static lv_obj_t * scr1;
static lv_obj_t * scr2;
static lv_obj_t * win;

static lv_res_t btn1_action(lv_obj_t * btn1)
{

    lv_obj_t * new_btn = lv_btn_create(win, NULL);

//  Or align to the top middle
    lv_obj_align(new_btn, NULL, LV_ALIGN_IN_TOP_MID, 0, 20);

    /*Add control button to the header*/
    lv_win_add_btn(win, SYMBOL_SETTINGS, NULL); /*Add a setup button*/

    /*Load scr2 to see what happened*/
    lv_scr_load(scr2);

   return LV_RES_OK;
}

void component_test(void)
{

    scr1 = lv_obj_create(NULL, NULL);
    scr2 = lv_obj_create(NULL, NULL);

    /*Initialize screen 1 with a button*/
    lv_obj_t * btn1 = lv_btn_create(scr1, NULL);
    lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, btn1_action);

   /*Create a window on screen 2*/
   win = lv_win_create(scr2, NULL);
   lv_win_set_title(win, "Example window (scr2)"); /*Set the title*/

   lv_obj_t * txt1 = lv_label_create(win, NULL);
   lv_label_set_text(txt1, "This is the content of the window\n\n"
   "You can add control buttons to\nthe window header\n\n"
   "You can scroll it\n\n"
   "See the scroll bar on the right!");

   lv_scr_load(scr1);
}
```
[image]

closed 2018-12-20 by chaos4284