lvgl/lvgl #643: issue with font

himsha, 2018-12-17:
Hi,
i am facing error in changing font style

this one is working
`    
    /*Create a new style for the label*/
    static lv_style_t style;
    lv_style_copy(&style, &lv_style_plain);
    style.text.color = LV_COLOR_BLUE;
    style.text.font = &lv_font_dejavu_20;
`

but when i change font from lv_font_dejavu_20 to lv_font_dejavu_10 its shows error
`  
    /*Create a new style for the label*/
    static lv_style_t style;
    lv_style_copy(&style, &lv_style_plain);
    style.text.color = LV_COLOR_BLUE;
    style.text.font = &lv_font_dejavu_10; // here its shows error
`

kisvegabor, 2018-12-17:
Hi,

Please copy here the error messasge?

BrianPugh, 2018-12-17:
is `lv_font_dejavu_10` enabled in your `lv_conf.h` ? Also try a `make clean`, just to be sure.

kisvegabor, 2018-12-18:
> 'lv_font_dejavu_10' undeclared (first use in this function)	welcome_screen.c 

It should be what @BrianPugh said. Enable the font in `lv_cong.h`

himsha, 2018-12-18:
i had declared as below, is this be fine? or how to enable it in lv_cong.h

LV_FONT_DECLARE(lv_font_dejavu_10);

kisvegabor, 2018-12-18:
You need `#define USE_LV_FONT_DEJAVU_10              4` in `lv_conf.h`. The number means the bit-per-pixels. It can be 1/2/4/8