lvgl/lvgl #644: Double frame buffer

szjdb, 2018-12-17:
Hello kisvegabor,
Thanks for your great lib!

    I got a question about the using the double frame buffer in F429. As there have some tearing effect when moving the page in high speed , it is causing by the conflict of the LTDC and the flush DMA working on the same buffer the same time. The better way is to write one frame buffer and left the other fetched by LTDC.
    Is there any arrangement for this kind of double buffering?
Best Regards,

kisvegabor, 2018-12-18:
Hi,

> Hello kisvegabor,
Thanks for your great lib!

It already not my lib, it's developed by a lot of great people! But thank you anyway, I happy to hear that you like it! :smile: 

> Is there any existing arrangement or machanism for this kind of double buffering?

Try `LV_VDB_DOUBLE` in `lv_conf.h`.

szjdb, 2018-12-18:
Hello kisvegabor,
Thanks for all!
    I had found such define in the lv_conf.h ,but I was  wondering that there is still lack of the synchronization between the LTDC and the VDB flush even with the help of double VDB. There have not any code to manage the LTDC interupt in the lib. We should change the frame buffer address in the LTDC interupt when the VDBA is ready, right?
    Further question about the VDB, I was planning to directlly draw the point on the VDB, but I found the VDB.area is changing according to the object, and I can't see the point I have draw. Is the VDB the map of the frame buffer? Anything wrong ? 
Would like to have your further advice! 
Best Regards,

szjdb, 2018-12-19:
Hello kisvegabor,
I got know the behavier of VDB. But still question about the way to use double frame buffer to cut the tearing effect. 
Would you please give some more advice?
Best Regards,

kisvegabor, 2018-12-19:
@szjdb 
The big question is how to trigger to tearing effect signal.  I found this the datasheet of SSD1963:
[image]

However, as far as I know, the LTDC module (I suppose you have an STM32 MCU) doesn't provide this signal. But it provides a VSYNC interrupt which gives the start of a frame. 
It's quite hacky but you can start a timer in the VSYNC interrupt and to get a Timer interrupt when the display is refreshed and you are in the "blank zone".

Would you like to change only teh frame buffer address or start a DMA transfer in the "blank zone"?

szjdb, 2018-12-19:
Hello kisvegabor,
I am using F429 with its LTDC. There will have LTDC interupt when in the VNDP peried to indicate the finish of one frame. If we can change the frame buffer addresss to the updated one in the interupt callback and let the lvgul drawing on the old one, there will be perfect.  The drawback of this mode is have to copy the changed area between each frame buffer and seems difficult to handle without the lib management.
Best Regards,

kisvegabor, 2018-12-20:
What is the resolution of your display?

We can add some options to the library to better support true double buffering (only address changing).
To do this you need two screen sized VDBs which will be the frame buffers too.

It should work something like this:
1. Set VDB1 as the current frame buffer
2. Copy VDB1 to VDB2
3. Render the image to VDB2 (only what is changed)
4. Set VDB2 as framebuffer 

Do you agree so far?

szjdb, 2018-12-21:
Hello kisvegabor,
Is it impossible to implement such double frame buffer solution in current VDB refreshing mechanism?

Best Regards,

kisvegabor, 2018-12-22:
Hi,

I reworked the VDB system a little bit to support double buffering using only address changing.

Here is how to use it:
1. In `lv_conf.h` set `LV_VDB_DOUBLE  1`
2. Set `LV_VDB_SIZE` to `LV_HOR_RES * LV_VER_RES`
3. If you place the frame buffers into external RAM set the addresses in `LV_VDB_ADR` and `LV_VDB2_ADR`
4. Register a rounder function:
 ```c

void round_cb(lv_area_t * a)
{
    a->x1 = 0;
    a->y1 = 0;
    a->x2 = LV_HOR_RES - 1;
    a->y2 = LV_VER_RES - 1;
}
...
    lv_refr_set_round_cb(round_cb);
``` 
5. In your `disp_flush` function change the address of the layer to `color_p` and call `lv_flush_ready()`

Now it redraws the whole screen even if you press only one button.

I also updated the `monitor` driver for PC to test it. Do the same settings in `lv_conf.h` (`LV_VDB(2)_ADR` can be `0`) and in `lv_drv_conf.h` add `#define MONITOR_DOUBLE_BUFFERED 1`.

Please let me know how it works.

kisvegabor, 2018-12-22:
I tested it on STM32F429 Discovery too. I was not synchronizing to VSYNC but framebuffer address changing was working.

szjdb, 2018-12-22:
Hello kisvegabor,
Thanks a lot!
I will test it soon!
Best Regards,

szjdb, 2018-12-23:
Hello kisvegabor,
I have done the step 1-4, except the 5. Then I tracked my code on disp_flush is calling , the color_p is allways point to the VDB1 and none change to VDB2.  
For the step 4, you means changing the frame buffer address instead of doing flush?

Best Regards,

kisvegabor, 2018-12-23:
Have you updated to dev-5.3 branch?

szjdb, 2018-12-23:
Hello kisvegabor,
Sorry, I am just using ver 5.2 . Is it a must?
Best Regards,

embeddedt, 2018-12-23:
@szjdb The VDB reworking necessary for double buffering was only applied to `dev-5.3`. However, I think the branch is stable enough that you should be able to use it, since we are planning to release `v5.3` in January.

kisvegabor, 2018-12-23:
@szjdb And `dev-5.3` is backward compatible. So you can update without any issues.

szjdb, 2018-12-23:
Hello kisvegabor and embeddedt ,
I just replaced the lib with the ver 5.3 , it is go to the hardfault when running lv_task_handler(). 
I am tracking the issue.

Best Regards and  Merry Christmas,

embeddedt, 2018-12-23:
@szjdb Where is it crashing? Possibly I can help you pinpoint the issue more quickly.

szjdb, 2018-12-23:
Hello embeddedt,
It is inside lv_task_handler(),when taking and running the new task ,it just jump to hardfault handler. I was running with DOUBLE VDB=0 right now and I also tested it with =1 with the same result .
Many thanks,

embeddedt, 2018-12-24:
@szjdb Sorry for the delay, didn't check my email. Can you provide a backtrace?

szjdb, 2018-12-24:
Hello kisvegabor and embeddedt,
Thanks a lot!
I found the porting issue in my TFT driver and now it is running. The address is change per every flush cycle now. I am trying to synchronize with the LTDC .
I understood the mechanism for the double VDB . It's really great but might be little bit costly to update the whole object in the active VDB  even just pressing one button. The better is to just update the changed object in the active VDB and copy them to the second VDB when changing the VDB?  

Best Regards and Merry Christmas Eve!

szjdb, 2018-12-25:
Hello kisvegabor and embeddedt, 
I found out that the bad quality is introduced by my vdb_wr function, so I turned back to use the default. 
As I had tested yestoday, it might have some further chance to speed up the refresh by just copy the redrawn object to the active VDB to keep the  synchronize of both VDB . How are your oppinion.

Best Regards  and Merry Christmas to you and your family, @kisvegabor @embeddedt

kisvegabor, 2018-12-26:
@szjdb Glad to hear that you made it work! :)

I was thinking about the speed up you described. Let's imagine the following use case:
- there are some bars to display some real-time measured data
- they are animated or changing frequently
- there is a  screen sized animation too

So what we can do?
1. Allocate two frame buffers in SRAM or as you want.
1. Set `LV_VDB_SIZE` to a smaller value, e.g. (`30* LV_HOR_RES`) and `LV_VDB_DOUBLE   0`. Set configure LittelvGL as "normal" one frame buffered mode.
2. In the `disp_flush` you will get only the changed areas and coordinates and you can copy them to your inactive frame buffer 
3. After some time (e.g. 50 ms) swap the frame buffers

However, the large screen sized animation will be drawn in more parts because you have only a small VDB. If you swap the frame buffers in the middle (not after the last part) you will get tearing effect. To avoid this you need to know when LittlevGL is still refreshing. There is no API for this if it would you could do something like this:
```c
if(is_swap_time_elapsed() && lv_refr_in_prog() == false)
{
   swap_frame_buffers();
   reset_swap_timer();
}
```

If you like this idea I can add the `lv_refr_in_prog()` function.

szjdb, 2018-12-28:
Hello kisvegabor，
I have thought about it these day and if the mechanism for  the double frame buffer  is too difficult , how about  just copy the drawn buffer to the drawing buffer at the beganning and refresh the one who needed, just like the emwin ?
May I have your oppinion?
Best Regards, @kisvegabor

szjdb, 2018-12-29:
Hello kisvegabor，  @kisvegabor 
Many thanks!
I will test it Asap!  For more detail, is there  any switch to turn on /of the ture double buffer mode in run time? 
 BTW, is there any posible to resize  the refresh area of the obj ? For example , a label on the button, is there any cb function to set to just refresh the label  area, not the whole button?  The purpose is also to speed up rendering. 
Best Regards,

kisvegabor, 2018-12-29:
> For more detail, is there any switch to turn on /off the ture double buffer mode in run time?

It's a compile-time setting now. Why would it be good to change it in run-time?

> BTW, is there any posible to resize the refresh area of the obj ? For example , a label on the button, is there any cb function to set to just refresh the label area, not the whole button?

Can you tell a concrete example?

> Also for the TA object, the whole area will be redrawn when just a cursor turning on/off. 

Yes, I know it and it should be fixed once. But not fixed yet...

szjdb, 2018-12-29:
Hello kisvegabor，
Thanks for so soon reply!
For the mode switch, the one VDB mode is the most steady and if there have issue with the double mode, we can switch back to the safty mode.
For the partrially refresh, one example is the chart object. If the seriers is update, the background is redrawn and then the whole colume, comparing to the mode just to erase or draw some part of colume element , it is really taking more time. That is also for the TA and if we can just erase or draw the cursor, the  time will be saving.
Best Regards,

kisvegabor, 2018-12-30:
@embeddedt @szjdb 

I fixed the Text area refresh issue. Can you test it?

kisvegabor, 2018-12-30:
@szjdb 

> For the mode switch, the one VDB mode is the most steady and if there have issue with the double mode, we can switch back to the safty mode in run time. That is just for in case.

I see. In v6.0 the display driver structure will be extended with more options because of multiple display support. True double buffering can be an option too. But for consistency, I'd like to leave it as compile time option until v6.0. 

 > For the partrially refresh, one example is the chart object. If the seriers is update, the background is redrawn and then the whole colume, comparing to the mode just to erase or draw some part of colume element , it is really taking more time. That is also for the TA and if we can just erase or draw the cursor, the time will be saving.

These are special cases. For example, it assumes, that there is no opacity on the background. LittlevGL searched to topmost fully covering object on an area and starts drawing from that object. These special drawings should be handled in `chart_design()` function. But it's a very different topic. If you wanna talk about this please open a new issue.

> Further extending to the object overlapping , if partrial drawing is capable , the hiden area of the bottom object could be drawn, even could been replaced by the saved hiden area, instead of redrawing the whole object when the top layer object is deleted. Have we implement some kind of mechanism like this?

I think it's not reasonable because of the huge memory impact.

> EDIT: I had just tested the true double VDB mode on the KB example. I set the LV_VDB_TRUE_DOUBLE_BUFFERED 1 and others as your advice, and just type some words in the TA window to make it scrollable , thus if I just scroll the TA, the KB should not be redrawn, but it does. Anything wrong with my setup?

Have you used `lv_test_kb_2()`? How do you know the Keyboard is also refreshed? It should not refresh Keyboard if you only scroll the Text Area.

szjdb, 2018-12-30:
Hello kisvegabor， @kisvegabor
Thanks for all!
I have tried,but the whole area is still refrshing. I just open the cursor and let it in LV_VDB_TRUE_DOUBLE_BUFFERED mode. Then I just track the lv_draw_rect(). It is still called and the coords is scaning from the top to the bottom of TA. There also have chance to draw the whole screen with the back ground color.

For the KB , I just track the lv_btnm_design() and it is still calling when no pressing on the TP and just the cursor is blanking.

For the partrially refreshing, another example is KB on an back ground picture , if one key is pressed , the whole KB will be refresh. If just refresh the pressed key area with back ground,  in despite of the OPA , it is also saving more.

May I have your further advice? 

Best Regards & Happy New Year,

kisvegabor, 2018-12-31:
For me `lv_btnm_design` is called on once with `lv_test_kb_1()`. 
Have you updated from `dev-5.3`?

To keep the issues clean let's talk about one topic in an issue.
So for partial refresh please open a new issue and discuss it there.

szjdb, 2018-12-31:
Hello kisvegabor， @kisvegabor
Thanks !
I have updated the the newest one as I can search the LV_VDB_TRUE_DOUBLE_BUFFERED usage in the project and none in the former . I am also sure the frame buffer is swapping form the two VDB.

May I have your further advice?
Best Regards & Happy New Year,

kisvegabor, 2018-12-31:
I've checked it's really uploaded. See: https://github.com/littlevgl/lvgl/blob/dev-5.3/lv_core/lv_vdb.c#L124

szjdb, 2018-12-31:
Hello kisvegabor， @kisvegabor
Thanks !
Yes I also check the lv_vdb.c and it is the same with your listed and the memcpy() has taken place. But as I said the lv_draw_rect() is still called and scanning. The high level lv_refr_obj is still calling the TA and KB design to refresh the whole object when none TP press.  That full object scanning design should stop when  none TP press, except the cursor blanking area?

Best Regards & Happy New Year,

kisvegabor, 2018-12-31:
@szjdb 
And the design functions are called with `mode == LV_DESIGN_DRAW_MAIN`?

I've tested it by adding a `printf` to `lv_ta_design`:
```c
static bool lv_ta_design(lv_obj_t * ta, const lv_area_t * mask, lv_design_mode_t mode)
{
    if(mode == LV_DESIGN_COVER_CHK) {
        /*Return false if the object is not covers the mask_p area*/
        return ancestor_design(ta, mask, mode);
    } else if(mode == LV_DESIGN_DRAW_MAIN) {
        /*Draw the object*/
        printf("mask: x1:%d, y1:%d, x2:%d, y2:%d\n", mask->x1, mask->y1, mask->x2, mask->y2);
        ancestor_design(ta, mask, mode);

    } else if(mode == LV_DESIGN_DRAW_POST) {
        ancestor_design(ta, mask, mode);
    }
    return true;
}
```
```
mask: x1:140, y1:30, x2:339, y2:129    <- first big draw
mask: x1:147, y1:38, x2:148, y2:58      <- 2px width cursor
mask: x1:147, y1:38, x2:148, y2:58
mask: x1:147, y1:38, x2:148, y2:58
mask: x1:147, y1:38, x2:148, y2:58
mask: x1:147, y1:38, x2:148, y2:58
mask: x1:147, y1:38, x2:148, y2:58
```

kisvegabor, 2018-12-31:
Aaand happy new year for you too :smile:

szjdb, 2019-01-01:
Hello kisvegabor， @kisvegabor
Thanks a lot !
Yes it is drawing on the mode LV_DESIGN_DRAW_MAIN. Your test show different result with mine. I was wondering if anything missing on my side.

Happy 2019 and Best Regards,

szjdb, 2019-01-01:
Hello kisvegabor， @kisvegabor
Thanks !
Yes I have tried the different configuration of the LV_VDB_TRUE_DOUBLE_BUFFERED and find almostly the same.
 In deep trace, I just find 3 major point what the LV_VDB_TRUE_DOUBLE_BUFFERED  concerned,  the lv_vdb_flush() ,lv_refr_task() and  lv_refr_area_with_vdb(). The former two are not the controlling of the refresh behavier ,and inside the last  one , there redefind the area of refreshing like:
    lv_vdb_t * vdb_p = lv_vdb_get();
    vdb_p->area.x1 = 0;
    vdb_p->area.x2 = LV_HOR_RES-1;
    vdb_p->area.y1 = 0;
    vdb_p->area.y2 = LV_VER_RES - 1;
    lv_refr_area_part_vdb(area_p);

Why do we refresh the whole area even in the LV_VDB_TRUE_DOUBLE_BUFFERED mode and no TP event input?
I had check the TA with the same lv_ta_ext_t definded with you.
Best Regards,

szjdb, 2019-01-01:
Hello kisvegabor， @kisvegabor
Thanks !
Yes, I was still keeping the lv_refr_set_round_cb(). That is the reason. When delete ,just the cursor is redrawn.Great!
Now can we specify the area to be copy to the active VDB, instead of the full screen?
Best Regards,

szjdb, 2019-01-03:
Hello kisvegabor， @kisvegabor
Thanks !
I means ,for further speeding up,  can we specify the area to be copy to the active VDB, instead of the full old VDB? Thus only the object which have been redrawn in the last active VDB should  be copied.
That will save more cpu load.
For example, in the KB example, we just copy the TA area, or even just a cursor area in the old VDB, to the active VDB. There will need a list of area to record the redrawn in the last VDB cycle.
Can we implement this?
Best Regards,

kisvegabor, 2019-01-04:
@szjb I added this. Now only the changed areas are updated in the new VDB.
Please test it!

szjdb, 2019-01-05:
Hello kisvegabor， @kisvegabor
Thanks a lot !
It's great! Only the changed area are copied even in animation! I am in deep testing.
BTW, is there any filelist box example in your website? 
Best Regards,

kisvegabor, 2019-01-05:
@szjdb 
Great! Waiting for the feedback!

> BTW, is there any filelist box example in your website?

Not yet. :(
If you are interested in implementing it, please open a new issue to discusse it.

szjdb, 2019-01-08:
Hello kisvegabor， @kisvegabor
Thanks a lot! Sorry for the delay.
Per the testing , it is perform good.
Further more, can we add some flag for the object which need not to be copy to the new VDB? As in the case of real time refreshing object like the curve , the old data is going to remove ,so to skip the copying is the best choice. 
May I have your advice?
Best Regards,

kisvegabor, 2019-01-08:
> Per the testing , it perform good.

Great!

> Further more, can we add some flag for the object which need not to be copy to the new VDB? As in the case of real time refreshing object like the curve , the old data is going to remove ,so to skip the copying is the best choice.

When LittlevGL refreshes it thinks about areas and not objects. Let's say the `(10;20); (100;130)` area needs to refreshed because a message box was there but it's just closed. LittlevGL will check which objects are there and draw their related parts in that area. When the objects are rendered the refreshed area is flushed and copied to the other frame buffer to keep it up to date. So at this point, LittlevGL doesn't know which objects were drawn there just see the resulted area and image. According to this the mentioned mechanism to not copy some objects is not possible :(

szjdb, 2019-01-08:
Hello kisvegabor， @kisvegabor
Many thanks ! 
I agree with you. What I mean is to set some kind of flag in the object /extend data to let the user control the copy action. It is normal to copy the refreshed area to new VDB ,but user could set it not to do by some kind of cb function. It is much help to the real time refreshed object, like the animation ,the fast chart and  the future mjpg,  as it will be redrawn frame by frame and it is just no means to copy the old area. The proper place of the flag might be in the lv_join_t of inv_buf[]?
May I have your futher advice?
Best Regards,

kisvegabor, 2019-01-09:
I see your point and it really would save some time. But as I mentioned when LittlevGL copies the areas it already doesn't know anything about the objects. For example:
1. A `chart` is marked as not-copy
2. When LittelvGL copes the areas can check all objects on the screen to see if some of them are on the area and have not-copy flag.
3. But what if a message box just appeared and overlays the `chart`. In this case, the area still needs to copied to see the message box on the other FB too.

Determining the order of the object, or the required truncation etc. would require more time then copy just the area.    

If you have a 300x200 chart = 60,000 px and refresh it 5 times/sec, LittlevGL will copy 300,000 px/sec. If 1 instruction needs to copy a px then with a 160MHz MCU it needs 1,8 ms.

embeddedt, 2019-01-09:
@kisvegabor @szjdb A 2 millisecond refresh time does not justify additional functionality, in my opinion. The human eye can't see that!

szjdb, 2019-01-09:
Hello kisvegabor，      @kisvegabor @embeddedt
Many thanks !
Yes the object overlay should be considered in deepth. In the case of some top object on a chart which will be refreshed frame by frame , to copy the area of the chart also have no means as the area will be redrawn in this frame together with the top object , per the exist refreshing rule. So not copying the marked area also help much as the chart often take most of the area of a screen. 
Further more , if some object  is marked as not-copy , the children and the overlay brother object also should not be copied , as they will be redraw in the new frame. This should have been involved in the existing refreshing rule.
May I have your futher advice?
Best Regards,

kisvegabor, 2019-01-10:
Let's say you marked as area as not-copy. Then LittlevGL must be sure that something will be drawn there in the next frame. So it can invalidate this area automatically. 
The area should be ignored only if there is an exact match among the redrawn area and not if e.g. the whole screen was redrawn. 

My only concern with this feature that is can be used on in true double buffered mode. Or can you imagine how can we use it other configurations too?

szjdb, 2019-01-11:
Hello kisvegabor， @kisvegabor 
Yes it will not copy the object area when it is invalid. But the invalid area should be defined by the object design function or the user cb. Otherwise , like in the TA, it will redraw the whole object again even just a cursor blanking. 
This feature is the only way to fix the tearing effect when in animation or live object without heavy load to the cpu. It seems very important for an up to date gui lib.
May I have your futher advice?
Best Regards,

szjdb, 2019-01-12:
Hello kisvegabor， @kisvegabor 
I agree with you in the above animation button example.  As the step of moving is usually small , copying the deleted button area is almost equal to the joined area of the new drawn button. So it need not to use the not-copy flag feature. 
But in the live chart and mjpeg case , it gain much because  saving of the large area copying, as they are usually take most part of the screen.  And the screen could be 800*600 and even larger.
For example, when a live is going to refresh:
1. lv_chart_set_points() to refresh the chart
2.  set the not-copying flag for the chart
3.  the chart is invalid and drawn in the active VDB1 and VDB2 is the displaying frame buffer
4.  change the frame buffer to VDB1
5. other drawn area except the chart are copied from VDB1 to the VDB2
6. go to the step1
When a message box going to pop on the chart 
1. lv_chart_set_points() to refresh the chart
2. set the not-copying flag for the chart
3. message box create check the overlay with the chart ,if does , also  set the not-copying flag for it
4. chart is drawn 1st then the message box  in the active VDB1 
4.  change the frame buffer to VDB1
5. other drawn area except the chart and the message are copied from VDB1 to the VDB2
6. go to the step1 and redraw the chart and the message box 

As the 50ms time slot and  other task the mcu need to run  , 1ms saving should be Considered.
Best Regards,

kisvegabor, 2019-01-16:
Sorry for the late answer. I was thinking about the opportunities.

> As the 50ms time slot sharing with multi task beside the GUI , 1ms saving should be Considered.

In my [previous calculation](https://github.com/littlevgl/lvgl/issues/644#issuecomment-452744507), the 2 ms gain was in 1 sec. But of course, it depends on the size of the area to copy, the speed of the MCU and the FPS.

I don't think that to check the objects again after drawing is a good way because there are a lot of questions and special cases here related to visibility, opacity, partial covering, animations etc. So it would be time consuming and error prone for some cases.

Instead of the `not-copy` flag, I can image an area (or an array of areas) to register in the display driver. These areas are considered to "continuously refreshed areas" that can be used for fast charts, movies etc. The library can invalidate and redraw these areas automatically in every frame. And in case of `TRUE_DOUBLE_BUFFERING` not copy these areas. It fits better into the library's working mechanism and seems more versatile to me.

kisvegabor, 2019-01-18:
My idea was to now copy an area at the end if it exactly matches with an area in the user-defined area array. It should happen in almost every cases. The exceptions are when e.g. a pop-up appears but it's only 1 frame (or a few with animation).

embeddedt, 2019-01-18:
@kisvegabor That sounds efficient to me as well.

kisvegabor, 2019-01-18:
@szjdb @embeddedt 

Then I leave this issue open as reminder for `v6.0`.

szjdb, 2019-01-19:
Hello kisvegabor， @kisvegabor
Many thanks!
So may I ask that when would the V6.0 go to release for testing?
Best Regards,

embeddedt, 2019-01-19:
@szjdb We're hoping to release `v6.0` in June. Development will start around the beginning of February.

szjdb, 2019-01-20:
Hello kisvegabor， @kisvegabor @embeddedt
Many thanks !
Can I ask for a temporay release for this improment as I am drawing a live chart with the lib and in tough timing. 
Best Regards,

embeddedt, 2019-01-20:
@szjdb You could use the `dev-6.0` branch before `v6.0` is officially released. However I'm not sure what order @kisvegabor wants to implement all of the features in. (We have quite a long list.)

kisvegabor, 2019-01-21:
@szjb As @embeddedt said it will be added to `dev-5.3` branch but I can't promise a deadline.
Until that, I suggest adding this feature yourself here: https://github.com/littlevgl/lvgl/blob/dev-5.3/lv_core/lv_refr.c#L229 Just not copy the area if its coordinates match with something you define.

[44 more replies]

closed 2019-03-15 by kisvegabor