lvgl/lvgl #645: clear last screen

himsha, 2018-12-18:
Hi,
after creating screen or menu do we need to clear or delete created label graph extra once created from jumping from one screen to other and how to do it?

kisvegabor, 2018-12-18:
Hi,

`lv_scr_load` won't delete anything. You can load the old screen again and it will have exactly the same content. Moreover, you can change objects even on the not loaded screen. E.g. create new objects or set size.

To save memory you can delete objects. But if you delete the whole screen, but be sure this is not the active screen. (Load a new screen before deleting one)

himsha, 2018-12-18:
any example?

kisvegabor, 2018-12-18:
Example of what exactly?

himsha, 2018-12-18:
how to remove header from tab?

embeddedt, 2018-12-18:
@himsha I will try and give some suggestions for each of the points you listed above.

> 1. splash screen (i had created but not able to erase it)

You should have a parent `lv_obj` that you set as the active screen with `lv_scr_act`. Then you can put your splash image, etc. on this `lv_obj`.

> 2. menu page (similar as tab demo available but tab header is not required as max space in screen is consumed by header and its buttons)

If you don't need the multi-page facilities of tabview then do something similar to what I wrote above. Otherwise @kisvegabor can probably tell you how to disable the header.

> 3. execution of each button available on menu (look similar to https://littlevgl.com/home/main_cover.png)

If you need a new screen for the button, again create a new `lv_obj` parent object to put all of the related widgets under.

himsha, 2018-12-19:
> @himsha I will try and give some suggestions for each of the points you listed above.
> 
> > 1. splash screen (i had created but not able to erase it)
> 
> You should have a parent `lv_obj` that you set as the active screen with `lv_scr_act`. Then you can put your splash image, etc. on this `lv_obj`.
> 
> > 1. menu page (similar as tab demo available but tab header is not required as max space in screen is consumed by header and its buttons)
> 
> If you don't need the multi-page facilities of tabview then do something similar to what I wrote above. Otherwise @kisvegabor can probably tell you how to disable the header.
> 
> > 1. execution of each button available on menu (look similar to https://littlevgl.com/home/main_cover.png)
> 
> If you need a new screen for the button, again create a new `lv_obj` parent object to put all of the related widgets under.

thanks

kisvegabor, 2018-12-19:
@himsha Unfortunately, there is no way to hide the button right now but you can apply this trick:
```c
    lv_obj_set_height(tabview, LV_VER_RES + 30);
    lv_obj_set_y(tabview, -30);
```

So just move the upper part out of the screen and adjust the height accordingly.

embeddedt, 2018-12-20:
@kisvegabor 

> Unfortunately, there is no way to hide the button right now...

From a quick read of `lv_tabview_create` this looks fairly simple to implement. I can send a PR if you want.

kisvegabor, 2018-12-20:
@embeddedt It would be very welcome!

kisvegabor, 2018-12-21:
@embeddedt Merged, thank you very much!

@himsha Please try this new feature and comment here is something is not working.

closed 2018-12-21 by kisvegabor