lvgl/lvgl #646: Full screen refresh times are getting longer and longer

DF-LeeX, 2018-12-19:
--time:21
--time:39
--time:57
--time:76
--time:95
--time:114
--time:133
--time:152
--time:171
--time:189
--time:208
--time:227
--time:247
--time:265
--time:284
--time:302
--time:323
--time:341
--time:360
--time:379
--time:398
--time:417
--time:434
--time:455
--time:473
--time:492

BesitzeRuf, 2018-12-19:
Increasing time value seems to be a constant (18-19)... Do you have code for sharing, to reproduce this problem?

embeddedt, 2018-12-20:
@DF-LeeX Can I ask why you are creating the same object multiple times? In most cases it should be suitable to initially create an object and just update it later (i.e. for `lv_label`, set the text to a new value with `lv_label_set_text`).

embeddedt, 2018-12-21:
@DF-LeeX What do you mean by dynamically brushing images? If you want to change the image displayed you can use `lv_img_set_src`.

kisvegabor, 2019-01-05:
@DF-LeeX 
As the original issue is solved I close it.
If the "dynamically brushing of images" issue is not solved yet, please open a new issue for it.

closed 2019-01-05 by kisvegabor