lvgl/lvgl #647: btn action callback function

MShahrukh0, 2018-12-19:
I need to add a function into button callback , and pass more than one lv_obj into that funcytion as arguments, i have gone through the documentation but couldn't find such function to set callback on button function and obj with that.
Can you tell me if it is possible what I am trying to achieve here?

kisvegabor, 2018-12-19:
I don't really understand. 

> I need to add a function into button callback , and pass more than one lv_obj into that funcytion

Is it your own function? Then it can have any prototype, like 
```c
void my_func(lv_obj_t * x, lv_obj_t * y)
{
   lv_obj_set_size(x, 20, 30);
   lv_obj_set_size(x, 60, 80);
}
```

embeddedt, 2018-12-20:
@kisvegabor I think what @MShahrukh0 might mean is that they want to be able to pass more than one `lv_obj` to their button callback. In that case maybe using `lv_obj_set_free_ptr` could help.

MShahrukh0, 2018-12-20:
yes @embeddedt  you are right that's exactly what i meant, can you please elaborate with the help of an small example?

kisvegabor, 2018-12-20:
In addition`free_ptr` can any data type, like 
```c
    static struct
    {
      lv_obj_t * obj1;
      lv_obj_t * obj2;
      lv_obj_t * obj3;
    }my_data;

    lv_obj_set_free_ptr(alloc_label, &my_data);
```

kisvegabor, 2019-01-05:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2019-01-05 by kisvegabor