lvgl/lvgl #648: Can not draw image in lvgl v5.2.

zaazbb, 2018-12-19:


zaazbb, 2018-12-19:
image struct:

```
lv_img_dsc_t img_qrcode = {
  .header.always_zero = 0,
  .header.w = 140,
  .header.h = 140,
  .data_size = 19600 * LV_COLOR_SIZE / 8,
  .header.cf = LV_IMG_CF_TRUE_COLOR,
  .data = img_qrcode_map,
};

```

my codes:

```
    LV_IMG_DECLARE(img_qrcode);

    lv_obj_t * img_symbol = lv_img_create(box2, NULL);
    lv_img_set_src(img_symbol, &img_qrcode);
```

zaazbb, 2018-12-19:
it's strange, now, it can draw image correctly.

zaazbb, 2018-12-19:
it can't draw again, disp "No data".

kisvegabor, 2018-12-19:
Hi,

I suppose you've created a custom `lv_img_dsc_t`.
In `lv_img_set_src` only a pointer to `src` is saved. So `img_qrcode` should global or static, but not a simple local variable.  

In addition, you can enable logging in `lv_conf.h`:
```c
/*Log settings*/
#define USE_LV_LOG      1   /*Enable/disable the log module*/
#if USE_LV_LOG

#define LV_LOG_LEVEL    LV_LOG_LEVEL_WARN
#define LV_LOG_PRINTF   1
#endif  /*USE_LV_LOG*/
```

If you don't have `printf` you can register a `print` function with `lv_log_register_print`.

zaazbb, 2018-12-20:
I create imq_qrcode use online image converter, and  those log were printed by lv_log already.
but the image still disp "No data".

- my image

, a [qrcode.]
[image]

the converter config.
[image]

- the img_qrcode.c:

```c
#include "lv_conf.h"
#include "lvgl/lv_draw/lv_draw_img.h"

const uint8_t img_qrcode_map[] = {
  0x89, 0x50, 0x4e, 0x47, 0x0d, 0x0a, 0x1a, 0x0a, 0x00, 0x00, 0x00, 0x0d, 0x49, 0x48, 0x44, 0x52, 
  0x00, 0x00, 0x00, 0x8c, 0x00, 0x00, 0x00, 0x8c, 0x08, 0x02, 0x00, 0x00, 0x00, 0x21, 0xa2, 0xd6, 
  0x69, 0x00, 0x00, 0x02, 0x86, 0x49, 0x44, 0x41, 0x54, 0x78, 0x9c, 0xed, 0xdd, 0xc1, 0x6e, 0xe3, 
  0x30, 0x0c, 0x00, 0xd1, 0xb8, 0xc8, 0xff, 0xff, 0xb2, 0x7b, 0xeb, 0x45, 0x06, 0x4a, 0x10, 0x24, 
  0xa5, 0x31, 0xe6, 0x1d, 0x17, 0xdb, 0x24, 0xee, 0x40, 0x00, 0x21, 0x57, 0xce, 0x75, 0xdf, 0xf7, 
  0x47, 0x67, 0xfb, 0xd9, 0xfd, 0x01, 0xf4, 0x3f, 0x23, 0x01, 0x18, 0x09, 0xc0, 0x48, 0x00, 0x46, 
  0x02, 0xf8, 0xae, 0xff, 0x74, 0x5d, 0xd7, 0xfc, 0xe7, 0xf8, 0x13, 0x99, 0x36, 0x83, 0x9f, 0x30, 
  0x37, 0xb8, 0x1e, 0x78, 0xf9, 0xae, 0x24, 0x00, 0x23, 0x01, 0x18, 0x09, 0xc0, 0x48, 0x00, 0x46, 
  0x02, 0x78, 0x98, 0xee, 0x56, 0x7d, 0xfb, 0x7b, 0x7b, 0x47, 0xa9, 0xa0, 0xed, 0x97, 0xef, 0x4a, 
  0x02, 0x30, 0x12, 0x80, 0x91, 0x00, 0x8c, 0x04, 0x60, 0x24, 0x80, 0xd0, 0x74, 0xb7, 0xca, 0x4d, 
  0x65, 0x55, 0x9b, 0x69, 0xc1, 0xd7, 0x89, 0x7c, 0xc8, 0xc9, 0xfd, 0xbd, 0xf4, 0x94, 0xe8, 0x4a, 
  0x02, 0x30, 0x12, 0x80, 0x91, 0x00, 0x8c, 0x04, 0x60, 0x24, 0x80, 0xe4, 0x74, 0xb7, 0x57, 0xeb, 
  0x9d, 0xd9, 0x03, 0xb9, 0x92, 0x00, 0x8c, 0x04, 0x60, 0x24, 0x00, 0x23, 0x01, 0x18, 0x09, 0x00, 
  0x39, 0xdd, 0xad, 0x63, 0xdb, 0xe3, 0xbc, 0xd7, 0xb7, 0x77, 0x37, 0xcc, 0x95, 0x04, 0x60, 0x24, 
  0x00, 0x23, 0x01, 0x18, 0x09, 0xc0, 0x48, 0x00, 0xc9, 0xe9, 0x6e, 0xef, 0x50, 0xd4, 0x7a, 0xaf, 
  0x36, 0x62, 0xf8, 0xf2, 0x5d, 0x49, 0x00, 0x46, 0x02, 0x30, 0x12, 0x80, 0x91, 0x00, 0x8c, 0x04, 
  0x10, 0x9a, 0xee, 0x4e, 0x3b, 0xfb, 0x10, 0xdc, 0xbb, 0xab, 0x9a, 0xc1, 0xb6, 0x5f, 0xbe, 0x2b, 
  0x09, 0xc0, 0x48, 0x00, 0x46, 0x02, 0x30, 0x12, 0x80, 0x91, 0x00, 0xae, 0xf3, 0x6f, 0x4d, 0xb6, 
  0x0e, 0x57, 0xe7, 0x5f, 0xfe, 0xc7, 0x95, 0x84, 0x60, 0x24, 0x00, 0x23, 0x01, 0x18, 0x09, 0xc0, 
  0x48, 0x00, 0xa1, 0xe7, 0xdd, 0x05, 0xf7, 0xca, 0x4a, 0xac, 0xef, 0x95, 0x7e, 0xf7, 0xdc, 0x0f, 
  0x4e, 0x5e, 0x5a, 0x90, 0x2b, 0x09, 0xc0, 0x48, 0x00, 0x46, 0x02, 0x30, 0x12, 0x80, 0x91, 0x00, 
  0xca, 0xf6, 0xee, 0x72, 0x33, 0x61, 0xd5, 0xb3, 0x8b, 0x87, 0x9f, 0x91, 0x52, 0xf5, 0x44, 0x94, 
  0xe0, 0x2f, 0xc4, 0x95, 0x04, 0x60, 0x24, 0x00, 0x23, 0x01, 0x18, 0x09, 0xc0, 0x48, 0x00, 0xa3, 
  0x67, 0x66, 0xab, 0xe6, 0xbd, 0xa0, 0xc9, 0x61, 0x32, 0xb7, 0x2b, 0x18, 0xbc, 0x58, 0x57, 0x12, 
  0x80, 0x91, 0x00, 0x8c, 0x04, 0x60, 0x24, 0x00, 0x23, 0x01, 0x18, 0x09, 0x20, 0xb9, 0xc1, 0xda, 
  0xb7, 0x9d, 0x5a, 0xf5, 0xee, 0xc3, 0x5a, 0x8f, 0x4c, 0xbb, 0x92, 0x00, 0x8c, 0x04, 0x60, 0x24, 
  0x00, 0x23, 0x01, 0x18, 0x09, 0x20, 0x34, 0xdd, 0x4d, 0xde, 0x1a, 0x1f, 0xde, 0x60, 0xdd, 0xfb, 
  0xe0, 0x42, 0x6f, 0x9f, 0xbf, 0x87, 0x91, 0x00, 0x8c, 0x04, 0x60, 0x24, 0x00, 0x23, 0x01, 0xbc, 
  0xe4, 0x60, 0xf3, 0xe3, 0x55, 0xf4, 0xfd, 0x09, 0x63, 0xee, 0x75, 0x56, 0x4e, 0x77, 0xef, 0x61, 
  0x24, 0x00, 0x23, 0x01, 0x18, 0x09, 0xc0, 0x48, 0x00, 0x0f, 0xd3, 0xdd, 0xde, 0x67, 0xf0, 0x15, 
  0x7e, 0x3b, 0x6f, 0x6e, 0x83, 0xb1, 0x6a, 0x3b, 0xb1, 0x70, 0x57, 0xd0, 0x95, 0x04, 0x60, 0x24, 
  0x00, 0x23, 0x01, 0x18, 0x09, 0xc0, 0x48, 0x00, 0xa1, 0xa3, 0x2f, 0x93, 0xb7, 0x26, 0x73, 0xff, 
  0xa7, 0xf0, 0x07, 0xab, 0x66, 0xc2, 0xaa, 0x57, 0xfe, 0xb8, 0x92, 0x10, 0x8c, 0x04, 0x60, 0x24, 
  0x00, 0x23, 0x01, 0x18, 0x09, 0x20, 0x79, 0xb0, 0xb9, 0xef, 0x66, 0x65, 0x4e, 0xfa, 0xb1, 0x35, 
  0xad, 0x53, 0x59, 0x15, 0x57, 0x12, 0x80, 0x91, 0x00, 0x8c, 0x04, 0x60, 0x24, 0x00, 0x23, 0x01, 
  0x8c, 0x3e, 0xb6, 0xa6, 0x4f, 0xe1, 0xd7, 0xc5, 0x55, 0xdd, 0xaa, 0x8e, 0x4c, 0x80, 0xde, 0x99, 
  0x7d, 0x0f, 0x23, 0x01, 0x18, 0x09, 0xc0, 0x48, 0x00, 0x46, 0x02, 0x40, 0x4e, 0x77, 0xe9, 0x9d, 
  0xba, 0xbe, 0x63, 0xbc, 0xad, 0x7f, 0xac, 0xe8, 0x4a, 0x02, 0x30, 0x12, 0x80, 0x91, 0x00, 0x8c, 
  0x04, 0x60, 0x24, 0x80, 0xe4, 0x74, 0x37, 0x79, 0xd2, 0x36, 0x7d, 0x63, 0xb4, 0x6a, 0xef, 0x6e, 
  0x3b, 0x57, 0x12, 0x80, 0x91, 0x00, 0x8c, 0x04, 0x60, 0x24, 0x00, 0x23, 0x01, 0x84, 0xa6, 0xbb, 
  0xbd, 0xa7, 0x68, 0x0b, 0xbf, 0x9c, 0x78, 0xef, 0xe9, 0x57, 0xbf, 0x50, 0xe4, 0xcd, 0x8c, 0x04, 
  0x60, 0x24, 0x00, 0x23, 0x01, 0x18, 0x09, 0x00, 0xf0, 0xbc, 0x3b, 0xb9, 0x92, 0x00, 0x8c, 0x04, 
  0x60, 0x24, 0x00, 0x23, 0x01, 0x18, 0x09, 0xc0, 0x48, 0x00, 0xbf, 0x88, 0x86, 0x03, 0x1d, 0x9a, 
  0x83, 0x2f, 0x86, 0x00, 0x00, 0x00, 0x00, 0x49, 0x45, 0x4e, 0x44, 0xae, 0x42, 0x60, 0x82, 
};

lv_img_dsc_t img_qrcode = {
  .header.always_zero = 0,
  .header.w = 140,
  .header.h = 140,
  .data_size = 703,
  .header.cf = LV_IMG_CF_RAW,
[... 559 lines cut ...]
Info: label created     (D:/work/repository/my/esp32/ESP32_CAMERA_QR-master/components/lvgl/lvgl/lv_objx/lv_label.c #129)
Info: container created         (D:/work/repository/my/esp32/ESP32_CAMERA_QR-master/components/lvgl/lvgl/lv_objx/lv_cont.c #108)
Info: image created     (D:/work/repository/my/esp32/ESP32_CAMERA_QR-master/components/lvgl/lvgl/lv_objx/lv_img.c #107)
Info: lv_img_set_src:  `LV_IMG_SRC_VARIABLE` type found         (D:/work/repository/my/esp32/ESP32_CAMERA_QR-master/components/lvgl/lvgl/lv_objx/lv_img.c #160)
Info: container created         (D:/work/repository/my/esp32/ESP32_CAMERA_QR-master/components/lvgl/lvgl/lv_objx/lv_cont.c #108)
[... cut ...]
Warn: Image draw error  (D:/work/repository/my/esp32/ESP32_CAMERA_QR-master/components/lvgl/lvgl/lv_draw/lv_draw_img.c #82)
Warn: Image decoder open: unknown color format  (D:/work/repository/my/esp32/ESP32_CAMERA_QR-master/components/lvgl/lvgl/lv_draw/lv_draw_img.c #460)
Warn: Image draw cannot open the image resource         (D:/work/repository/my/esp32/ESP32_CAMERA_QR-master/components/lvgl/lvgl/lv_draw/lv_draw_img.c #311)
Warn: Image draw error  (D:/work/repository/my/esp32/ESP32_CAMERA_QR-master/components/lvgl/lvgl/lv_draw/lv_draw_img.c #82)

```

kisvegabor, 2018-12-20:
If you would like to send so much text, next time,  for better readability please attach them as files instead.

Why did you choose *Raw* color format? 
For a normal image *True color* should be chosen.

zaazbb, 2018-12-21:
I don't know why, now use True color, it's ok.

closed 2018-12-21 by zaazbb