lvgl/lvgl #649: window color

himsha, 2018-12-19:
hi,
is is possible change window background color?

himsha, 2018-12-19:
done and as below

>     /*Create a parent object*/
    lv_obj_t * window = lv_obj_create(lv_scr_act(), NULL);    	/*Create a parent object on the current screen*/
    lv_obj_set_size(window, 480, 272);                       	/*Set the size of the parent*/
    lv_obj_set_style(window, &lv_style_plain);
    lv_obj_set_pos(window, 0, 0);

    lv_obj_t * notification_window = lv_obj_create(window, NULL);   /*Create a parent object on the current screen*/
    lv_obj_set_size(notification_window, 480, 20);                 /*Set the size of the parent*/
    lv_obj_set_pos(notification_window, 0, 0);

    /*Create an object and set a user create style for it*/
	static lv_style_t style;
	lv_style_copy(&style, &lv_style_pretty);
	style.body.main_color = LV_COLOR_BLACK;
	style.body.grad_color = LV_COLOR_BLACK;
	style.body.radius = 0;
	style.body.border.color = LV_COLOR_WHITE;
	style.body.border.width = 0;
	style.body.opa = LV_OPA_COVER;
	style.body.shadow.width = 0;
	lv_obj_set_style(notification_window, &style);

closed 2018-12-19 by himsha

kisvegabor, 2018-12-19:
Hi,

Great! However the intended way is:  `lv_win_set_style(win, LV_WIN_STYLE_BG, &style);`