lvgl/lvgl #650: XPT2046 pins connection??

miki105, 2018-12-19:
Hi,
I'm trying to find documentation about the hardware connections, between 2.8" ILI9341 display with driver touch XPT2046 and esp32 wroom32 that is used by your libraries, but after few days I decide to ask to you if you could post those information.
many thanks

kisvegabor, 2018-12-19:
Hi,

do you mean this project? https://github.com/littlevgl/lv_projects/tree/master/esp32_ili9341

If yes, unfortunately, the touchpad is not added to this project yet. :(
If you are interested in and would like to add it I'd kindly help

miki105, 2018-12-20:
Hi Gabor,
yes, I meant that project
If you could help it will be very kind. I have seen on your libraries under lv_drivers\indev that, XPT2046 is implemented but there are not IO map.
many thanks for the support.

miki105, 2018-12-20:
Could you do for me or it's a long time job for you?, or could you send an example how to do it.
Sorry I'm beginner on your libraries.

Many thanks

kisvegabor, 2018-12-21:
You just need to use [this file](https://github.com/littlevgl/lv_drivers/blob/master/indev/XPT2046.c) and replace the SPI related things with ESP SPI functions.

miki105, 2018-12-22:
I'm stuck at begin again, no IO map ,however
Many thanks.

miki105, 2018-12-24:
Hi Gabor,
many thanks really appreciated.
Merry Christmas

InfiniteYuan1, 2019-01-03:
@miki105 You can refer [iot-solution](https://github.com/espressif/esp-iot-solution/tree/master/examples/hmi), there are some example with xpt2046 on esp32

kisvegabor, 2019-01-31:
I've added the driver to https://github.com/littlevgl/esp32_ili9431

If it's not working properly, please open an issue in its repo.

closed 2019-01-31 by kisvegabor