lvgl/lvgl #651: Limited ammount of lv_list object

MichaelVLV, 2018-12-19:
Hi.

I'm trying to create list with 100 objects inside.
Problem:

When ammount of items is more than XX, I got crash:

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandl
ed.
Core 0 register dump:
PC      : 0x4013dd89  PS      : 0x00060730  A0      : 0x80144058  A1      : 0x3f
fe6970
0x4013dd89: lv_obj_create at C:/Projects/lvgl/lv_cor
e/lv_obj.c:1405

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc7ed0  A5      : 0x00
000014
A6      : 0x0000051d  A7      : 0x00000000  A8      : 0x8013dd87  A9      : 0x3f
fe6950
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffc7f60  A13     : 0x00
000001
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00
00001d
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xff
ffffff

Backtrace: 0x4013dd89:0x3ffe6970 0x40144055:0x3ffe6990 0x40144fdd:0x3ffe69b0 0x4
014446f:0x3ffe69d0 0x400d6c86:0x3ffe6a00 0x400d6fab:0x3ffe6a60 0x401468b5:0x3ffe
6a80 0x40146927:0x3ffe6aa0 0x400d705f:0x3ffe6ac0
0x4013dd89: lv_obj_create at C:/Projects/lvgl/lv_cor
e/lv_obj.c:1405

0x40144055: lv_cont_create at C:/Projects/lvgl/lv_ob
jx/lv_cont.c:67

0x40144fdd: lv_btn_create at C:/Projects/lvgl/lv_obj
x/lv_btn.c:57

0x4014446f: lv_list_add at C:/Projects/lvgl/lv_objx/
lv_list.c:342

0x400d6c86: avo_scr_history_menu at C:/Projects/main
/lvgl_tft.c:1279

0x400d6fab: avo_scr_switch_task at C:/Projects/main/
lvgl_tft.c:1279

0x401468b5: lv_task_exec at C:/Projects/lvgl/lv_misc
/lv_task.c:232

0x40146927: lv_task_handler at C:/Projects/lvgl/lv_m
isc/lv_task.c:232

0x400d705f: lvgl_tft_task at C:/Projects/main/lvgl_tft.c:1368 (discriminator 1)

```

My display is 160*128, part of my code (how does I create this list):

```C
void my_list()
{
	lv_obj_t * scr = lv_obj_create(NULL, NULL);
	lv_scr_load(scr);

	char buf[50];

	lv_obj_t * label;
	label = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_text(label, "My_list");
	lv_obj_align(label, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);

	/*Crate the list*/
	lv_obj_t * list1 = lv_list_create(lv_scr_act(), NULL);
	lv_obj_set_size(list1, 140, 70);
	lv_obj_align(list1, NULL, LV_ALIGN_IN_TOP_MID, 0, 30);

	/*Add list elements*/
	for (uint8_t i = 1; i < 60; i++)  // 60 - ok;   65 - lv_task freeses after some scrolling; 70 - instant crash when trying load screen
	{
		sprintf(buf, "list#%d", i);
		lv_obj_t * list_btn = lv_list_add(list1, NULL, buf, my_action);
		lv_btn_set_fit(list_btn, false, false);
		lv_obj_set_size(list_btn, 120, 18);
	}

	lv_group_add_obj(scr1_group, list1);

	lv_list_set_sb_mode(list1, LV_SB_MODE_AUTO);
}
```

kisvegabor, 2018-12-19:
Hi,

Try to increase `LV_MEM_SIZE` in `lv_conf.h`

MichaelVLV, 2018-12-20:
I have changed to 18KB (it was 16KB), now I'm able to load about 70. 
But if changed to higher values (more than 19KB), my FreeRTOS scheduler cant start...
Does this list really requires so much memory?

kisvegabor, 2019-01-10:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2019-01-10 by kisvegabor