lvgl/lvgl #652: about label set pos.

3038922, 2018-12-20:
My lvgl version is 5.1.0.
I have some problems with the location of LABEL.
[image]
[image]
[image]
[image]

embeddedt, 2018-12-20:
@3038922 Could you highlight where you are calling the sort functions? I am having troubles finding it even though I read your screenshots multiple times.

3038922, 2018-12-20:
[image]
I am a novice, thank you very much. I use C++ STL. to create objects.

kisvegabor, 2018-12-21:
@3038922 What is the parent of the label? If it's a page (or window) have a look at this: https://docs.littlevgl.com/#Page (Notes section)

3038922, 2018-12-21:
[image]

kisvegabor, 2018-12-22:
`lv_tabview_add_tab()` also craetes a page.

3038922, 2018-12-23:
> `lv_tabview_add_tab()` also craetes a page.
> 
> If still not working please provide a minimal example to see what happens and tell what you expect to happen.

Thank you for your patience. We are VEX users of educational robots. We come from all over the world. We use LVGL in the open source free IDE PROS. We are not professional programmers. We are all junior high school and high school students. So we ask amateur questions. Because of language problems, documents are often not carefully read. I'm very sorry.

kisvegabor, 2018-12-23:
Thank for the introduction! 

If you are interested, you can contribute to the library by translating the documentation.

Anyway, I suppose you were able to solve the issue using the documentation. If it's still not working, comment here.

closed 2018-12-23 by kisvegabor