lvgl/lvgl #654: LV_MEM_CUSTOM: DMA Problem

seyyah, 2018-12-20:
:key: Keywords: dma, usart, peripheral, stm32, LV_MEM_CUSTOM, memory leak

We use Stm32 boards with USART+DMA+Periphereal. There is an interesting memory problem.

Below minimal code fragment stops the DMA.

```c
static lv_obj_t * lbl = NULL;
char buf[5] = "1234567"; // overflow problem!!!

lbl = lv_label_create(lv_scr_act(), NULL);
lv_label_set_text(lbl, buf);
```

We fixed "overflow problem" and set `#define LV_MEM_CUSTOM      0` then the same behaviour/overflow arises. DMA stops.

We suspect the LV_MEM_CUSTOM produces memory leak.

Notes,
- LittlevGL v5.2
- lv_conf.h: https://gist.github.com/seyyah/f6deddd16b75ff1251887f2f7a81f145

BesitzeRuf, 2018-12-20:
Try to rewrite the line:
`char buf[5] = "1234567"; // overflow problem!!!`

by this:
`const char* buf = "1234567";  `

seyyah, 2018-12-20:
Thanks @BesitzeRuf for quick response. Our main problem is not `buf's overflow` but DMA stopping. We don't know real source/origin of problem.

We also suspect hashmap[1] as a source of DMA stopping problem.

## :clipboard: Reference
1. https://github.com/petewarden/c_hashmap

kisvegabor, 2018-12-20:
@seyyah 

AFIAK it's unpredictable what happens when this kind of overflow happens. It even can stop the DMA somehow. 

Occasionally, I run `lv_test_stress_1()` from `lv_examples` to test against memory leak. It seems working in `dev-5.3` and `v5.2` too.

seyyah, 2018-12-20:
@kisvegabor 

We use 10.4inch (800x600 pixel, 16 bit / RGB565) resolution display, What do you think about [1]

- #define LV_MEM_SIZE    (64U * 1024U) 
- #define LV_VDB_SIZE     (30 * LV_HOR_RES)

## :clipboard: Reference
1. Gist: https://gist.github.com/seyyah/f6deddd16b75ff1251887f2f7a81f145

upbeat27, 2018-12-21:
@seyyah only you can determine what is a good size for LV_MEM_SIZE. For example... in my case even 8k is more than enough.
Use `lv_mem_monitor` to see how much of LV_MEM_SIZE you are using on your most strenuous menu, in the worst possible scenario. 
After that you can determine how large it needs to be. Better to start large and work your way down once everything else is finalized.

kisvegabor, 2018-12-21:
@seyyah 
Regarding to `LV_MEM_SIZE` I agree with @upbeat27 . You should use `lv_mem_monitor` to see the memory usage of your GUI.

Regarding to `LV_VDB_SIZE` now you have 1/20 screen sized buffer. My experience is from 1/20 to 1/5 there isn't a big difference. If you would use less VDB (let's say 1/100) the library needed to check everything a lot of times. E.g. check all objects on the screen, check if they are needed to be drawn or not, start drawing algorithms severs times etc. So giving a larger VDB will reduce this overhead, however, this overhead is small. Doing the process 5 or 10 or 20 times doesn't make big difference.

seyyah, 2018-12-24:
@upbeat27 and @kisvegabor thank you for `sysmon_create` trick. Our memory consumption is below `10%` for 64MB and approximately 80% for 8MB. So it requires big memory for we have large display. 64MB (LV_MEM_SIZE) enough for our scenerio.

Our screen resolution is 800x600 pixels. We select LV_VDB_SIZE as `600/10 * LV_HOR_RES`.

## :mag: Details
Our test code in pc_simulator and screen froze for 8MB.

Phase 1:
```c
   lv_test_stress_1();
```

The screen froze. Console log is

```sh
pc_simulator$ ./demo 
used:   3224 ( 40 %), frag:   0 %, biggest free:   4968
used:   4480 ( 55 %), frag:   1 %, biggest free:   3696
used:   4632 ( 57 %), frag:   1 %, biggest free:   3544
used:   5504 ( 68 %), frag:   5 %, biggest free:   2576
used:   6224 ( 76 %), frag:   3 %, biggest free:   1920
used:   6160 ( 76 %), frag:   6 %, biggest free:   1920
used:   6160 ( 76 %), frag:   6 %, biggest free:   1920
Warn: Couldn't allocate memory  (lvgl/lv_misc/lv_mem.c #145)
Error: Out of memory!   (lvgl/lv_core/lv_obj.c #187)
```

Phase 2:

```c
   lv_test_stress_1();
   sysmon_create();
```

Console log is

```sh
used:   5504 ( 68 %), frag:   0 %, biggest free:   2688
used:   6760 ( 83 %), frag:   2 %, biggest free:   1416
used:   6912 ( 85 %), frag:   2 %, biggest free:   1264
used:   7784 ( 96 %), frag:  28 %, biggest free:    296
Warn: Couldn't allocate memory  (lvgl/lv_misc/lv_mem.c #145)
Error: Out of memory!   (lvgl/lv_core/lv_obj.c #187)
```

Our lv_conf.h for phase 1 and phase 2

```c
#define LV_MEM_SIZE    (8U * 1024U)
#define LV_HOR_RES          (800)
#define LV_VER_RES          (600)
```

## :camera: Screen shots
[image]

embeddedt, 2018-12-24:
@seyyah You have `LV_MEM_SIZE` set to 8 **KB** not MB.

seyyah, 2018-12-24:
@embeddedt I'm sorry for typo: ~8MB~ -> 8KB

#define LV_MEM_SIZE    (8U * 1024U)

embeddedt, 2018-12-24:
@seyyah What exactly is the problem? From your report above it sounds like you know that you need to increase `LV_MEM_SIZE`.

BesitzeRuf, 2018-12-24:
> Is there a known problem in the use of LittlevGL with previously experienced DMA?

No any. There very good examples from [lv_projects](https://github.com/littlevgl/lv_projects).. Did you try run just simple demo example on your hardware (without use of any other library or third party code)?

seyyah, 2018-12-24:
We're just going to test a simple example containing (USART+) DMA on the STM32F746G-disco. We will share our experience. Thanks @BesitzeRuf

kisvegabor, 2019-01-10:
@seyyah is there any news in this topic?

seyyah, 2019-01-10:
Unfortunately there is not any good news. We had to change peripheral DMA to Polling. Thanks

closed 2019-01-10 by seyyah