lvgl/lvgl #658: about group focus

longdadouzi, 2018-12-21:
Can I hide the focus of the group on the object when I switch to another object group?

kisvegabor, 2018-12-22:
I've pushed an update to `dev-5.3` to hide the focused object with `lv_group_set_style_mod_cb(g, NULL);`

If you use the default style modifier function you should get it before setting it to `NULL` with `lv_group_get_style_mod_cb`.

If you don't want to use `dev-5.3` branch you can create a dummy hidden (`lv_obj_set_hidden(dummy_obj, true)`) and focus on it.

longdadouzi, 2018-12-24:
Ok, understand

longdadouzi, 2018-12-24:
Upgrade version, is it compatible with lower version functions?

embeddedt, 2018-12-24:
@longdadouzi For the most part, yes. If there is a compilation error let us know and we will help you resolve it.

longdadouzi, 2018-12-26:
Can I remove the color when the object group is focused? And you can use the controls. When you focus on the roller control, the color is not too good to look at.

longdadouzi, 2019-01-25:
I think I have mastered it and can close this question.

closed 2019-01-25 by kisvegabor