lvgl/lvgl #660: ddlist_open argument names are inconsistent between .h and .c file

rreilink, 2018-12-21:
How to name the second argument of lv_ddlist_open? For the Python bindings, the name of the argument matters (since one could call `ddlist.open(anim_en = True)`)

lv_ddlist.c:
```
void lv_ddlist_open(lv_obj_t * ddlist, bool anim_en)
```

lv_ddlist.h:
```
/**
 * Open the drop down list with or without animation
 * @param ddlist pointer to drop down list object
 * @param anim_en true: use animation; false: not use animations
 */
void lv_ddlist_open(lv_obj_t * ddlist, bool anim);
```

kisvegabor, 2018-12-22:
I reviewed this `bool` parameter names and updated them to be more consistent. Now if the function name clearly shows that what the `bool` parameter does it's just called `en`. 

I did it manually and checked only the `bool ..._en` parameters. So probably there are more inconsistencies too.

rreilink, 2018-12-23:
I may be able to use the parser that I am using for generating bindings, to check consistency between h and c (between function declaration and definition)

kisvegabor, 2018-12-23:
@rreilink It'd be great!

kisvegabor, 2018-12-30:
The other inconsistencies are shown in #681

closed 2018-12-30 by kisvegabor