lvgl/lvgl #662: icon label

himsha, 2018-12-22:
hi,
how to put label under button?

embeddedt, 2018-12-22:
@himsha Create a label object parented to the button's parent rather than the button itself. Then use `lv_obj_align` to align the label object under the button object.

embeddedt, 2019-01-01:
@himsha Are you referring to the apps list on a smartphone?

kisvegabor, 2019-01-01:
@himsha You can use `layouts`:
1. Create a parent which supports `layout` like `lv_cont` and set `lv_cont_set_layout(bg, LV_LAYOUT_GRID)`
2. Create an `lv_cont` parent for the button and text and set `lv_cont_set_layout(button_cont, LV_LAYOUT_COL_M)`

Learn mor about layouts here: https://docs.littlevgl.com/#Container

himsha, 2019-01-03:
> @himsha You can use `layouts`:
>
> 1. Create a parent which supports `layout` like `lv_cont` and set `lv_cont_set_layout(bg, LV_LAYOUT_GRID)`
> 2. Create an `lv_cont` parent for the button and text and [...]

hi i had don as below, when ever i add new button it get automatically in row, how to put buttons in a grid? 

	lv_obj_t * box1;
	box1 = lv_cont_create(lv_scr_act(), NULL);
	lv_obj_set_style(box1, &lv_style_pretty);
	lv_cont_set_fit(box1, true, true);
	lv_cont_set_layout(box1, LV_LAYOUT_GRID);

	/*Create a normal button*/
	lv_obj_t * btn1 = lv_btn_create(box1, NULL);
	lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, btn_click_action);

	lv_obj_t * btn2 = lv_btn_create(box1, NULL);
	lv_btn_set_action(btn2, LV_BTN_ACTION_CLICK, btn_click_action);

kisvegabor, 2019-01-03:
Try this:
```c
lv_obj_t * box1;
    box1 = lv_cont_create(lv_scr_act(), NULL);
    lv_obj_set_style(box1, &lv_style_pretty);
    lv_cont_set_layout(box1, LV_LAYOUT_GRID);
    lv_obj_set_size(box1, 400, 300);

    uint8_t i;
    for(i = 0; i < 6; i++) {
        lv_obj_t * holder = lv_cont_create(box1, NULL);
        lv_cont_set_fit(holder, true, true);
        lv_cont_set_layout(holder, LV_LAYOUT_COL_M);

        lv_obj_t * btn1 = lv_btn_create(holder, NULL);
        lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, NULL);

        char buf[32];
        lv_obj_t * label = lv_label_create(holder, NULL);
        sprintf(buf, "Button %d", i);
        lv_label_set_text(label, buf);

    }
```
[image]

turoksama, 2019-01-04:
@kisvegabor 
Hi,
 Can make that cont clickable with styles like btn? Or should we add a feature like that?

kisvegabor, 2019-01-04:
@turoksama 
You can just use a button instead.

closed 2019-02-01 by stale[bot]