lvgl/lvgl #663: find error

caoliuchao, 2018-12-23:
Hello! I am a college student from China. I am really glad to find littleVGL, which is a very good GUI. However, when I studied littleVGL, I found some problems. In version 5.2, the return value of "lv_arc_get_angle_end(lv_obj_t * arc)" function in the source code is" ext->angle_start", maybe there is something wrong with the source code, maybe I don't understand it. I hope you can tell me.

embeddedt, 2018-12-23:
@caoliuchao Good catch! `ext->angle_end` is definitely used in other places like `lv_arc_set_angles` so it belongs in `lv_arc_get_angle_end` as well. This problem is in `v5.3` as well.

closed 2018-12-23 by embeddedt

kisvegabor, 2018-12-23:
@caoliuchao Thank you for pointing this out!
And @embeddedt Thank you for the fix :)

kisvegabor, 2018-12-23:
@embeddedt 
The notification above says: 
[image]

How did you close a PR in a commit? Or did I miunderstand something?

embeddedt, 2018-12-23:
@kisvegabor When you put `Fixes #<issue-number>` in a commit, it will automatically close the related issue once the commit is pushed to `master`. I didn't make a PR for this change.

https://help.github.com/articles/closing-issues-using-keywords/

kisvegabor, 2018-12-23:
@embeddedt I see, thanks! Good to know. It's a really useful feature!

commit 6f5c1ba6 (2018-12-23): Fix wrong return value of lv_arc_get_angle_end (Closes #663)