lvgl/lvgl #665: How to design a modifiable config,like Spinbox?

3038922, 2018-12-23:
My LVGL version is 5.1.0.
I don't have Spinbox OBJ;
I want to design a button that functions like Spinbox, but I have no idea.
I use c++ vector to create automatically according to the number of SD card objects.
But I can't specify the action for each upDownBtn.

[image]

#include "ncrapi/system/systemData.hpp"

static lv_res_t saveAction(lv_obj_t *btn)
{
    (void)btn; /*Unused*/
    bool flag = sysData->saveConfig(sysData->_pidData, "/usd/pid.txt");
    if (!flag)
        std::cerr << "pid data save unsuccessful" << std::endl;
    return LV_RES_INV;
}
static lv_res_t upDownAction(lv_obj_t *btnm, const char *txt)
{
    (void)btnm; /*Unused*/
    //TODO 做一个字符串判断
    if (!strcmp(txt, "U"))
        sysData->setConfig(sysData->_pidData, "", "");
    if (!strcmp(txt, "D"))
        sysData->setConfig(sysData->_pidData, "", "");
    return LV_RES_INV;
}

namespace ncrapi
{
void UserDisplay::createPidTest(lv_obj_t *parent)
{

    if (displayObj[OBJ_BTNM_SON] == nullptr)
        displayObj[OBJ_BTNM_SON] = lv_tabview_create(parent, nullptr);

    lv_tabview_set_style(displayObj[OBJ_BTNM_SON], LV_TABVIEW_STYLE_BTN_REL, &btnStyle); //设置样式
    lv_obj_t *frTab = lv_tabview_add_tab(displayObj[OBJ_BTNM_SON], "前后PID调试");
    lv_obj_t *rotateTab = lv_tabview_add_tab(displayObj[OBJ_BTNM_SON], "左右PID调试");
    static const char *updown[] = {"U", "D", ""};
    std::vector<lv_obj_t *> labs;
    std::vector<lv_obj_t *> values;
    std::vector<lv_obj_t *> updownBtn;
    for (auto &it : sysData->_pidData) //很精髓的创建方式
    {
        if (it.first.find("前后_") != std::string::npos)
        {
            labs.push_back(lv_label_create(frTab, nullptr));
            values.push_back(lv_label_create(frTab, nullptr));
            updownBtn.push_back(lv_btnm_create(frTab, nullptr));
        }
        else
        {
            labs.push_back(lv_label_create(rotateTab, nullptr));
            values.push_back(lv_label_create(rotateTab, nullptr));
            updownBtn.push_back(lv_btnm_create(rotateTab, nullptr));
        }
        if (labs.size() != 1 && labs.size() != 5)
            lv_obj_align(labs.back(), *(labs.end() - 2), LV_ALIGN_OUT_BOTTOM_LEFT, 0, 20); //页面创建的内容排序有问题
        lv_btnm_set_map(updownBtn.back(), updown);
        lv_obj_set_size(updownBtn.back(), 60, 25);
        lv_obj_align(values.back(), labs.back(), LV_ALIGN_OUT_RIGHT_MID, 60, 0);
        lv_obj_align(updownBtn.back(), values.back(), LV_ALIGN_OUT_RIGHT_MID, 30, 0);
        lv_btnm_set_action(updownBtn.back(), upDownAction);
        lv_label_set_static_text(labs.back(), it.first.c_str()); //设置静态文本
        lv_label_set_text(values.back(), it.second.c_str());
    }

    //创建保存按钮
    lv_obj_t *saveBtn = lv_btn_create(displayObj[OBJ_BTNM_SON], nullptr);
    lv_obj_set_pos(saveBtn, LV_HOR_RES - 140, LV_VER_RES - 50);
    lv_obj_set_size(saveBtn, 50, 25);
    lv_obj_t *saveLab = lv_label_create(saveBtn, nullptr);
    lv_label_set_text(saveLab, "save");
    lv_btn_set_action(saveBtn, LV_BTN_ACTION_CLICK, saveAction);

    //退出按钮
    createExitBtn(OBJ_BTNM_SON);
}

} // namespace ncrapi

embeddedt, 2018-12-23:
@3038922 Exactly what is not working? Is there a compile-time error or does the action just not work?

embeddedt, 2018-12-23:
@3038922 Which objects do you mean by `corresponding address of each object`? The `btnm` pointer passed to the action tells you which button matrix the action was invoked on.

Please explain exactly what you are trying to do.

3038922, 2018-12-24:
#include "ncrapi/system/systemData.hpp"

static lv_res_t saveAction(lv_obj_t *btn)
{
    (void)btn; /*Unused*/
    bool flag = sysData->saveConfig(sysData->_pidData, "/usd/pid.txt");
    if (!flag)
        std::cerr << "pid data save unsuccessful" << std::endl;
    return LV_RES_INV;
}

static lv_res_t upDownAction(lv_obj_t *btnm, const char *txt)
{
    (void)btnm; /*Unused*/
    lv_obj_t *labs = static_cast<lv_obj_t *>(lv_obj_get_free_ptr(btnm));
    lv_obj_t *val = static_cast<lv_obj_t *>(lv_obj_get_free_ptr(labs));

    float temp = sysData->read<float>(sysData->_pidData, lv_label_get_text(labs));
    if (!strcmp(txt, "U"))
        temp += 0.001;
    if (!strcmp(txt, "D"))
        if (temp >= 0.001)
            temp -= 0.001;

    std::ostringstream oss; //主要为了去掉小数点后多余的0 默认3位
    oss << temp;
    sysData->setConfig(sysData->_pidData, lv_label_get_text(labs), oss.str());
    lv_label_set_text(val, oss.str().c_str());

    return LV_RES_INV;
}

namespace ncrapi {
void UserDisplay::createPidTest(lv_obj_t *parent)
{

    if (displayObj[OBJ_BTNM_SON] == nullptr)
        displayObj[OBJ_BTNM_SON] = lv_tabview_create(parent, nullptr);
    lv_obj_t *frTab = lv_tabview_add_tab(displayObj[OBJ_BTNM_SON], "前后PID调试");
    lv_obj_t *rotateTab = lv_tabview_add_tab(displayObj[OBJ_BTNM_SON], "左右PID调试");
    static const char *updown[] = {"U", "D", ""};
    std::vector<lv_obj_t *> labs;

    std::vector<lv_obj_t *> values;
    std::vector<lv_obj_t *> updownBtn;
    for (auto &it : sysData->_pidData)
    {
        if (it.first.find("前后_") != std::string::npos)
        {
            labs.push_back(lv_label_create(frTab, nullptr));
            values.push_back(lv_label_create(frTab, nullptr));
            updownBtn.push_back(lv_btnm_create(frTab, nullptr));
        }
        else
        {
            labs.push_back(lv_label_create(rotateTab, nullptr));
            values.push_back(lv_label_create(rotateTab, nullptr));
            updownBtn.push_back(lv_btnm_create(rotateTab, nullptr));
        }
        lv_obj_set_free_ptr(updownBtn.back(), labs.back()); 
        lv_obj_set_free_ptr(labs.back(), values.back());    
        if (labs.size() != 1 && labs.size() != 5)
            lv_obj_align(labs.back(), *(labs.end() - 2), LV_ALIGN_OUT_BOTTOM_LEFT, 0, 20);

embeddedt, 2018-12-24:
@3038922 Glad to hear it worked!

If your problem is solved please close this issue.

closed 2018-12-24 by 3038922