lvgl/lvgl #667: platformio support

puzrin, 2018-12-24:
[platformio](https://platformio.org/) is quite popular for embedded developmend, because is solves a lot of everyday tasks (dependencies install, firmware upload, build for multiple platforms)

It would be nice to support it, instead of manual copying lvgl to project folder.

kisvegabor, 2018-12-25:
Hi @puzrin 

I didn't know _platformio_ so far but it seems like a great and popular IDE. I agree, adding LittlevGL to it would be a good idea.

Do you experience with adding libraries to _platformio_?

embeddedt, 2018-12-25:
@puzrin What do you mean by hardcoded config path? lv_conf.h can be placed anywhere.

puzrin, 2018-12-25:
I mean now config expected to be near `lvgl` folder https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_arc.h#L20.

Alternative as placing it into `path` seems a bit imprecise.

embeddedt, 2018-12-26:
@puzrin What is imprecise about LV_CONF_INCLUDE_SIMPLE? I think we can leave it to the user not to be placing multiple identically-named copies of lv_conf.h in their project.

puzrin, 2018-12-26:
Well, i'm not experienced with C, and don't know the best practices for such situations. If you think `LV_CONF_INCLUDE_SIMPLE` is optimal as main use case, then  next step is to add lib into pio registry and create demo for vscode.

puzrin, 2018-12-28:
@kisvegabor i'm NOT using lvgl right now (but would like to use it with ease). Example of working project with platformio https://github.com/speedcontrols/ac_sc_grinder.

Here is short idea:

1. You need keep `/.vscode` folder, that will force vscode to install platformio plugin on project open.
2. After platformio installed, it will use `platformio.ini` to install dependencies and tools.

So, you can tell to anyone: "just install vscode, open folder and wait until it downloads everything needed".

First time you have to install platformio manually, see instruction https://platformio.org/platformio-ide. Feel free to ask anything, i'll try to help.

kisvegabor, 2018-12-31:
@puzrin 
Thanks for the idea!

I tried it with my STM32F429 Discovery but wasn't able to compile my project because some definitions were missing. I'm not sure but seems the `stm32f4xx_hal_..._ex` files are missing. 

Do you use a dev board or a custom board? If dev board which one?

puzrin, 2018-12-31:
I don't understand you. If you ask about [this](https://github.com/speedcontrols/ac_sc_grinder) project, it's for real device on custom PCB, all instructions and additional links are in readme.

If you mean, how to combine CubeMX with PlatformIO - use [config](https://github.com/speedcontrols/ac_sc_grinder/blob/master/platformio.ini) as start point. Also see [here](https://github.com/speedcontrols/ac_sc_grinder/tree/master/lib/stm32cubemx_init), how to tweak cubemx files and forget about this nightmare :)

kisvegabor, 2018-12-31:
> I don't understand you. If you ask about this project, it's for real device on custom PCB, all instructions and additional links are in readme.

Sorry, I somehow I missed the link to your project above. 

> If you mean, how to combine CubeMX with PlatformIO - use config as start point. Also see here, how to tweak cubemx files and forget about this nightmare :)

So you are generating a project with CumeMX first, and then add PIO config files manually? 
I've created a project with VSCode and selected CubeMX as the framework. But if I understood well it is not the best way :)

puzrin, 2018-12-31:
You are right, code, generated by CubeMX is far from perfect. But we had no time to learn how to properly init everything manually, add freertos and so on... So, we combined kludges to move forward :).

Initially, we planned to write code in Rust. And C code was considered as "temporary mess" and nobody cared too much about specific details. We only cared about architecture and data flow.

kisvegabor, 2018-12-31:
Ahh, I see. Meanwhile, I found that with PIO, from STM32F4 family, I can create only generic STM32F407 project which has only 3 SPI. So there is no chance for making it work with the default configuration. 
I will follow what you suggested! Thank you!

kisvegabor, 2018-12-31:
I installed your fixmath library to see how libraries are working in PIO and it was working well. Great lib, by the way! 

And I registered LittelvGL with this [library.json](https://raw.githubusercontent.com/littlevgl/lv_projects/master/platformio/library.json) file! So I'm waiting to appear in the list and test it. 

Now, I've just copied everything into to `src` folder and to make it work I only need to add the `src` folder as include path:
```
src_build_flags =
  -I src
```

[image]

kisvegabor, 2019-01-01:
I was looking for an `/inc` folder but it wasn't there by default and I supposed it is not used in PIO.
Anyway, it would be better to place the header files there. I will update it.

Starting the PC simulator with Eclipse on Linux is easy, but on Windows always there are challenges. And it needs the SDL library to be installed externally. However, I agree, an easy-to-get-started simulator for PIO would be great! I will think about it.

I found that PIO is like a better Arduino targeting the same group of people. Its professional IDE, debugging, great MCU/Board support really makes a difference compared to Arduino. So it's very nice. 

By the way, I didn't find how to add examples to a library. Do you know how to do it?

puzrin, 2019-01-01:
We can discuss simulator separate, if you wish. For example, could it be done in browser, without additional libs at all, cross platform. As far as i understand, you need to resend screen rectangle and get mouse/kbd back. If you are ok about frames drop, that's doable. I could try to help about JS part.

> By the way, I didn't find how to add examples to a library. Do you know how to do it?

PIO examples (for registry) are not executable, that's just a useless snippets. If you need executable examples in the same repo - they could use lvgl via local relative path.

- published library uses `library.json`
- "examples" (pio project?) uses `platform.ini`

So, no conflict.

> I found that PIO is like a better Arduino targeting the same group of people. Its professional IDE, debugging, great MCU/Board support really makes a difference compared to Arduino. So it's very nice.

I'd say, the root of problem is, that c/c++ has no standard dependencies management (cross platform package installer) and users are not motivated to follow some distribution conventions. PIO tries to fix that, but that's still a kludge, and ecosystem will never bee good.

I have to write code "right now", and have to select C because no immediate alternative available. But as soon as i can, i will switch to Rust. It already works, but need some time until people implement popular libraries for embedded. You can see survey for 2019 [here](https://github.com/rust-embedded/wg/issues/256).

kisvegabor, 2019-01-01:
> We can discuss simulator separate, if you wish. For example, could it be done in browser, without additional libs at all, cross platform. As far as i understand, you need to resend screen rectangle [...]

An online simulator would be awesome. It's my old-old dream... :) Basically, LittlevGL needs a canvas where the rendered image can be copied and get back the mouse coordinates and button state. 
Would it mean a JS binding? Or C parsed by JS?

> PIO examples (for registry) are not executable, that's just a useless snippets.

I just want to add a "useless" snippet with some basic instructions.

> I have to write code "right now", and have to select C because no immediate alternative available. But as soon as i can, i will switch to Rust. It already works, but need some time until people implement popular libraries for embedded. You can see survey for 2019 here.

I was surprised to see that Rust among the ten most popular languages. We are working on [Python/Micropython binding to LittelvGL](https://github.com/littlevgl/lvgl/issues/557). AS it will be the first binding LittlevGL will be updated to better support binding generation. I can imagine a Rust binding too in the future.

puzrin, 2019-01-01:
> An online simulator would be awesome. It's my old-old dream... :) Basically, LittlevGL needs a canvas where the rendered image can be copied and get back the mouse coordinates and button state.
Would it mean a JS binding? Or C parsed by JS?

I meaned a bit different thing:

- application with "emulator" backend creates webserver and respond with html/js
- user point browser to application page, then page open websocket connection and listen for canvas refreshes.

"emulator" should have screen buffer inside, and be ready to send it every 20-100ms when listener attached.

This may look strange and dirty, but should work everywhere.

> I just want to add a "useless" snippet with some basic instructions.

I did not used this feature. May be this is option you need: https://docs.platformio.org/en/latest/librarymanager/config.html#examples

puzrin, 2019-01-02:
https://github.com/novnc/noVNC - may be useful for info. VNC-based solution, how to stream screen to browser.

Modern solution are based on HTML5 video tag & WebRTC. May be more complicated to embed server side part.

kisvegabor, 2019-01-03:
Hi,

Thank you for the suggestions!

We were talking about Emscripten support in this thread[this thread](https://github.com/littlevgl/lvgl/issues/687) and @embeddedt already made it work. I think it's a very simple solution. 

I uploaded the first test here: https://littlevgl.com/em/lvgl.html

What do you think about it?

puzrin, 2019-01-03:
> What do you think about it?

I think it's very amazing as presentation, and very useful to promote lvgl.

If we speak about development - as "stupid user" i need simple simulator target install. For example, via add 1 dependency to PIO config. If you can provide such solution - it's not principle what technology is used behind.

kisvegabor, 2019-01-04:
> If we speak about development - as "stupid user" i need simple simulator target install. For example, via add 1 dependency to PIO config. If you can provide such solution - it's not principle what technology is used behind.

Yes, it would be great but I still didn't find and easy to install, cross-platform GUI library for PC. 
The PC simulator uses SDL which is sometimes difficult to install on Windows. We just need to open a window with a canvas or texture and to put the rendered image there. Do you have any idea for this?

puzrin, 2019-01-04:
For "1-click install" everything should be included in simulator backend module.

My idea was "embedding server with websockets should be more simple" (and use browser/s canvas). Have you seen https://github.com/novnc/noVNC for available server side libs?

embeddedt, 2019-01-04:
@kisvegabor 

> The PC simulator uses SDL which is sometimes difficult to install on Windows.

Once I clean up the Windows drivers, we could set up `pc_simulator` to use it if it is running on Windows.

puzrin, 2019-01-04:
After googling a bit SDL alternatives, found this link https://www.khronos.org/opengl/wiki/Related_toolkits_and_APIs. People mention GLFW & SFML. But i'm not "C programmer", and have no ideas is it more simple to embed those or not. At least, there are no more native libs to search.

Currently, i see 2 potential possibilities of simulator implementations:

- Use native libraries (SDL/GLFW/SFML/...)
- Use browser via embedded websockets/vnc/WebRTC server

Also, i don't promote idea of "vendor lock" to PlatrofmIO. But if it allows "1-click simulator backend install" - that's "better than nothing". If we care about easy development, then manual install is not good.

kisvegabor, 2019-01-05:
@puzrin 
> Use browser via embedded websockets/vnc/WebRTC server

Now I pay only for a webserver to host littlevgl.com. Can it work on this?

> Use native libraries (SDL/GLFW/SFML/...)

I have no experience with that but maybe compiling SDL too can work. So by building SDL to the target system can provide compatibility. But I don't how complicated it is to build SDL.

@embeddedt 
> Once I clean up the Windows drivers, we could set up pc_simulator to use it if it is running on Windows.

Having a "zero-configuration" driver at least for Windows would be really helpful! I'll add it once you are ready!

puzrin, 2019-01-05:
> Now I pay only for a webserver to host littlevgl.com. Can it work on this?

I can not imagine scenario when this can help. Start point is application, end point is browser. If you wish to place littlevgl.com between, that will be overcomplication IMHO.

kisvegabor, 2019-01-05:
> I can not imagine scenario when this can help. Start point is application, end point is browser. If you wish to place littlevgl.com between, that will be overcomplication IMHO.

I'm not a web developer, so maybe I don't cleary understand the concept but `WebRTC` needs a remote server, right?

kisvegabor, 2019-01-05:
Or `localhost` but it also needs some kind of installation.

puzrin, 2019-01-05:
> I'm not a web developer, so maybe I don't cleary understand the concept but `WebRTC` needs a remote server, right?

No, that's just a UDP-based protocol. If you need http pages - you should embed "http server". If you need html5 video - you need to embed "webrtc server". Just another kind of library.

> Or `localhost` but it also needs some kind of installation.

Sure, you will have to add some library sources (with tcp stack, http / websockets / webrtc / vnc serbers) to PC sim. That may be more simple, than add SDL sources. Or may be not :). I don't know. At least, this doesn't look mad.

kisvegabor, 2019-01-05:
@puzrin 
I see. Thank you for the explanation. Unfortunately, I haven't got enough experience in web development to implement it in a reasonable time. Are you interested in contributing with a driver like this?

puzrin, 2019-01-05:
I'd prefer to be useful with something different. Personally, i don't like C and use it only when absolutely no choice available.

kisvegabor, 2019-01-05:
I see. Thank you for the suggestion anyway. Let's is somebody would like to give it a try!
Until that, I will add @embeddedt's Windows driver. It should help a lot of people to get started with a few clicks. @puzrin which OS do you use?

puzrin, 2019-01-05:
> which OS do you use?

Linux. Ubuntu 18.04 lts.

kisvegabor, 2019-01-05:
Then at least in your case using SDL should be easy. https://docs.littlevgl.com/#PC-simulator

kisvegabor, 2019-01-05:
By the way, I'm just trying to run the SDL project in PIO. But I can't create a PC project? Do you know how to do it?

puzrin, 2019-01-06:
> By the way, I'm just trying to run the SDL project in PIO. But I can't create a PC project? Do you know how to do it?

Create [one more tagret](https://github.com/speedcontrols/ac_sc_grinder/blob/876ceb6da7efec30f04db6fce3f7e6af89d6b8a9/platformio.ini#L34) with `native` platform.

https://docs.platformio.org/en/latest/platforms/native.html

kisvegabor, 2019-01-06:
Thank you! I will try with it!

kisvegabor, 2019-01-06:
I've checked the docs and your example. 

I can compile it but when I tried `Debug/Start without Debugging` I got `Error: Please setup environments in `platformio.ini` file`

platformio.ini:
```
[platformio]
description = Nothing...
env_default = test_native

[env:test_native]
platform = native
lib_ignore = stm32cubemx_init
```
Do you have any idea?

puzrin, 2019-01-06:
Also, you can ask in parallel at https://community.platformio.org/, they are very responsive.

kisvegabor, 2019-01-07:
Thank you! 
I've found the solution in the readme: 

```
# Run program
> .pioenvs/native/program
```
So the program needs to be started from the terminal instead of the menu. It worked this way.

I added the  SDL driver and the whole project works now. LittelvGL is installed as a library.

[image]

puzrin, 2019-01-07:
Sounds nice. Do you plan to publish SDL driver in PIO registry? IMHO it would be convenient, even if SDL should be installed manually.

kisvegabor, 2019-01-08:
I agree and will add the drivers soon.

kisvegabor, 2019-01-23:
Finally, I added the drivers and create a project which should work is SDL is already installed:
https://github.com/littlevgl/pc_simulator_sdl_platformio

puzrin, 2019-01-23:
Awesome! IMHO, main goal of this issue is solved - library can be used without manual src copy, git submodules and other boring things.

Note about "tutorial". It would be very helpeful to have example with dual targeting (native and one of stm32/esp32 board with embedded oled). To understand how to properly split platform-dependend code. May be (not sure 100%), that will require to use RTOS to hide most of details about delays.

Such example could be good initial project template for beginners. If you wish - i can create new issue about (to let this one be closed and solved).

kisvegabor, 2019-01-24:
> Awesome! IMHO, main goal of this issue is solved - library can be used without manual src copy, git submodules and other boring things.

I think so!

> Note about "tutorial". It would be very helpeful to have example with dual targeting (native and one of stm32/esp32 board with embedded oled). To understand how to properly split platform-dependend code. May be (not sure 100%), that will require to use RTOS to hide most of details about delays.

Good idea. It should be easy. Please open an new issue for it.

puzrin, 2019-06-20:
https://community.platformio.org/t/platformio-plus-goes-open-source-improving-embedded-development-community-worldwide/8240

PIO opensourced testing and debugging parts. No more limits!

embeddedt, 2019-06-20:
@puzrin Awesome! This should definitely increase its popularity.

kisvegabor, 2019-06-21:
Nice! It makes platform.io a real alternative.