lvgl/lvgl #669: About button double-click settings

longdadouzi, 2018-12-25:
I can't seem to find the button on the data.
Excuse me, when I press the button, do not release the button, can I set the callback function for the continuous call button?

embeddedt, 2018-12-25:
@longdadouzi It's `LV_BTN_ACTION_LONG_PR_REPEAT`. See https://docs.littlevgl.com/#Button.

longdadouzi, 2018-12-26:
I have tried this and it can't trigger any callback function. Is it necessary to set anything?

embeddedt, 2018-12-26:
@longdadouzi What are the LV_INDEV settings in lv_conf.h?

longdadouzi, 2018-12-27:
In addition to using LV_BTN_ACTION_CLICK, the other can not make the callback function valid, such as LV_BTN_ACTION_LONG_PR_REPEAT, are invalid
[image]

embeddedt, 2018-12-27:
@longdadouzi Do any action callbacks work?

longdadouzi, 2018-12-27:
NO, only valid for the callback of LV_BTN_ACTION_CLICK, otherwise invalid, this may be the reason why the double button cannot be pressed.
@embeddedt

longdadouzi, 2018-12-28:
The return value is false because the buffer is not used, but the double button cannot be used.

MShahrukh0, 2018-12-28:
@longdadouzi when your button then in struct input_event you'll get different values , e.g,
struct input_event in;
//if button pressed you'll get in.value = 1,
//if button released you'll get in.value = 0;
//if button is long pressed you'll get in.value = 2;
if you want to make your button work in a long pressed condition then use in.value == 2 condition,
you'll find this in evdev.c file

longdadouzi, 2018-12-28:
As you said, how to set the button to press and continue to work; I can't seem to find a way to set it, except that LV_BTN_ACTION_CLICK other settings are invalid. Need me to set a different flag, instead of simply setting the press or release.
@MShahrukh0

longdadouzi, 2018-12-28:
Do you mean that I need to explore this file and try to change it so that it supports continuous callbacks?
@MShahrukh0

MShahrukh0, 2018-12-28:
@longdadouzi I faced same issue like you are facing right now, it is sad that only LV_BTN_ACTION_CLICK works, so what I am saying that you add your callback on this , but  read key when it is in long pressed mode or pressed mode or release mode.
And for that you need to go in evdev.c file and customize it accordingly.

embeddedt, 2018-12-28:
@MShahrukh0 @longdadouzi As far as I know LittlevGL checks the time your input device is pressed automatically. So all you need to know at the time of `indev_read` is whether the device is pressed or released. You don't need to keep track of the long-press state yourself.

MShahrukh0, 2018-12-29:
@embeddedt @longdadouzi I am using it in long press mode, actually when we add LV_BTN_ACTION_CLICK action then it mean a click is a combination of press+release , so in click case when we press and release then the button click works, when we press the button for long then we get input event value as 2 , so we can do as follow,
`//just pass NULL in place of call back
lv_btn_set_action(btn, LV_BTN_ACTION_CLICK,NULL);
bool evdev_read(lv_indev_data_t * data)
{
 struct input_event in;

    while(read(evdev_fd, &in, sizeof(struct input_event)) > 0) {
if(in.type == EV_KEY) {
if(in.value == 0){
data->state = LV_INDEV_STATE_REL;
data->key = keycode_to_ascii(in.code);
printf("Released key code 2 %d\n",in.code);
}
else if(in.value == 1){
data->state = LV_INDEV_STATE_PR;
data->key = keycode_to_ascii(in.code);
printf("Pressed key code 2 %d\n",in.code);
}
else if(in.value == 2){
//here you can declare your callback function
if(in.code == yourKeyCode)
{
btn_callback();
}
printf("Long pressed key code 2 %d\n",in.code);
}
}`

kisvegabor, 2018-12-30:
@MShahrukh0 
What about? 
```c
if(in.value == 0){
				data->state = LV_INDEV_STATE_REL;
				data->key = keycode_to_ascii(in.code);
				printf("Released key code 2 %d\n",in.code);
			}
			else if(in.value == 1 || in.value == 2){
				data->state = LV_INDEV_STATE_PR;
				data->key = keycode_to_ascii(in.code);
				printf("Pressed key code 2 %d\n",in.code);
			}
}
```

Or have look at this implementation: https://github.com/littlevgl/lv_drivers/blob/master/indev/keyboard.c

kisvegabor, 2018-12-30:
According to this `in.value == 1 || in.value == 2` can work: 
```
Event: time 1546152042.138907, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.138907, type 1 (EV_KEY), code 30 (KEY_A), value 1
Event: time 1546152042.138907, -------------- SYN_REPORT ------------
aEvent: time 1546152042.391524, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.391524, type 1 (EV_KEY), code 30 (KEY_A), value 2
Event: time 1546152042.391524, -------------- SYN_REPORT ------------
aEvent: time 1546152042.431788, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.431788, type 1 (EV_KEY), code 30 (KEY_A), value 2
Event: time 1546152042.431788, -------------- SYN_REPORT ------------
aaEvent: time 1546152042.472096, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.472096, type 1 (EV_KEY), code 30 (KEY_A), value 2
Event: time 1546152042.472096, -------------- SYN_REPORT ------------
aaEvent: time 1546152042.516412, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.516412, type 1 (EV_KEY), code 30 (KEY_A), value 2
Event: time 1546152042.516412, -------------- SYN_REPORT ------------
aEvent: time 1546152042.556795, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.556795, type 1 (EV_KEY), code 30 (KEY_A), value 2
Event: time 1546152042.556795, -------------- SYN_REPORT ------------
aaEvent: time 1546152042.596936, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.596936, type 1 (EV_KEY), code 30 (KEY_A), value 2
Event: time 1546152042.596936, -------------- SYN_REPORT ------------
aaEvent: time 1546152042.637225, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.637225, type 1 (EV_KEY), code 30 (KEY_A), value 2
Event: time 1546152042.637225, -------------- SYN_REPORT ------------
aEvent: time 1546152042.678535, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.678535, type 1 (EV_KEY), code 30 (KEY_A), value 2
Event: time 1546152042.678535, -------------- SYN_REPORT ------------
aEvent: time 1546152042.709357, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
Event: time 1546152042.709357, type 1 (EV_KEY), code 30 (KEY_A), value 0
Event: time 1546152042.709357, -------------- SYN_REPORT ------------

```

turoksama, 2018-12-30:
> According to this `in.value == 1 || in.value == 2` can work:
>
> ```
> Event: time 1546152042.138907, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1e
> Event: time 1546152042.138907, type 1 (EV_KEY), code 30 [...]

Hi，
How to print these messages on screen in SIMULATOR debug stage?

kisvegabor, 2018-12-30:
I've used [evtest](http://manpages.ubuntu.com/manpages/trusty/man1/evtest.1.html).

MShahrukh0, 2018-12-31:
> @MShahrukh0
> What about?
>
> ```c
> if(in.value == 0){
> 				data->state = LV_INDEV_STATE_REL;
> 				data->key = keycode_to_ascii(in.code);
> 				printf("Released key code 2 %d\n",in.code);
> 			}
> 			else [...]

@kisvegabor this may work actually.....  only if user want same output on both action click as well as long press, because in my code I am using one button for more than one function that's why i needed to separate each in.value.

kisvegabor, 2018-12-31:
@MShahrukh0 
LittlevGL will measure the length of the pressing and call the corresponding button action. So different actions will be called for a short click and long press. See: https://docs.littlevgl.com/#Button

kisvegabor, 2019-01-15:
From this code snippet, it seems you are using the `master` branch. This part of the code was already updated in `dev-5.3`. Please switch to `dev-5.3` branch.

longdadouzi, 2019-01-16:
It seems to find the problem, why it needs LV_GROUP_KEY_ENTER to enter a long press operation. And I use UP or Down to enter a long press operation.

MShahrukh0, 2019-01-16:
@longdadouzi you can redefine function on each key, in my code i am not using LV_GROUP_KEY_ENTER for enter function.

kisvegabor, 2019-01-16:
@longdadouzi 
It's because `LV_GROUP_KEY_ENTER` is used to replace the "press" operation on the touchpad. `LV_GROUP_KEY_LEFT/RIGHT` is used to set the object's value.  What is your exact use case when you need "long press" action on right/left press?

kisvegabor, 2019-01-22:
Happy to hear that!

closed 2019-01-22 by kisvegabor