lvgl/lvgl #670: Christmas lights controller using littlevgl, ESP32, MQTT

mharizanov, 2018-12-25:
Hello,
I'd like to share  a project I completed using littlevgl, ESP32 and my own built hardware: a Christmas lights controller with touch, MQTT:
[image]

The hardware is ILI9341+touch+ESP32, read more about it here: https://harizanov.com/wiki/wiki-home/tft32/

Source code is available here for you to try: https://github.com/mharizanov/TFT32/blob/master/examples/gui/mqtt_controller/mqtt_controller.ino

Thanks to Gabor & team for making this library available!

mfisher-azafran, 2018-12-25:
Very nice.  Thanks for sharing.

kisvegabor, 2018-12-26:
Hi @mharizanov,

Thank you for sharing your project!  :+1: 
I've just created a blog post where we can collect the projects with LittlevGL.
Here it is: https://blog.littlevgl.com/2018-12-26/references

If you want, please add your project to the list!

mharizanov, 2018-12-27:
Neat,
thanks, I did that. 
Cheers

closed 2018-12-27 by mharizanov