lvgl/lvgl #675: How to add lvgl to existing Makefile project

MrMoszk, 2018-12-27:
I'm trying lvgl with stm32 mcu. I have a cubemx generated Makefile project and have included lvgl.
Why do the *.mk files use variable LVGL_DIR for CFLAGS only ? But not for VPATH or DEPPATH

right now i have this in my makefile. Is this the right way to add lvgl to makefile project?

```
LVGL_DIR = Drivers
CFLAGS += -IDrivers/lvgl
CFLAGS += -DLV_CONF_INCLUDE_SIMPLE
include $(LVGL_DIR)/lvgl/lv_core/lv_core.mk
include $(LVGL_DIR)/lvgl/lv_draw/lv_draw.mk
include $(LVGL_DIR)/lvgl/lv_fonts/lv_fonts.mk
include $(LVGL_DIR)/lvgl/lv_hal/lv_hal.mk
include $(LVGL_DIR)/lvgl/lv_misc/lv_misc.mk
include $(LVGL_DIR)/lvgl/lv_objx/lv_objx.mk
include $(LVGL_DIR)/lvgl/lv_themes/lv_themes.mk
C_SOURCES += $(CSRCS)
VPATH := $(subst lvgl,$(LVGL_DIR)/lvgl,$(VPATH))
DEPPATH := $(subst lvgl,$(LVGL_DIR)/lvgl,$(DEPPATH))
```

kisvegabor, 2018-12-28:
Hi,

Good observation! 

The current Makefile was designed to work with NuttX but it seems your updates won't break it. Using absolute path seems event better. 
I updated `VPATH` and `DEPPATH` in the `.mk` files according to your suggestion and I will update the `Makefile` in the v5.3 release too (in a few weeks)

kisvegabor, 2019-02-01:
Updated here: https://github.com/littlevgl/pc_simulator_sdl_eclipse/blob/master/Makefile

Thank you for the suggestion!

closed 2019-02-01 by kisvegabor