lvgl/lvgl #678: LCD Sheild Pins

autoprog, 2018-12-28:
Hi, my test with 429-Disc1 +ILI9341 ok,
now, have two Sheilds, ILI9341 and SSD1963, for test RT1021 NXP.

my question 1: this shield does not have the  pins  **LCD_TFT HSYNC  | LCDTFT VSYNC | LCD_TFT DE**  
how to ?

[image]

[image]

Question 2:  DMA, have tutorial how to config ?  (my case Cortex-M7 ).
Question 3: RAM start and size ?

thanks
Carlos.

kisvegabor, 2018-12-30:
Hi,

> my question 1: this shield does not have the pins LCD_TFT HSYNC | LCDTFT VSYNC | LCD_TFT DE
how to ?

These shields are designed to use with a parallel port but the ILI9341 on the Discovery board is configured to drive with an LCD (RGB) interface. So you should use the FSMC (parallel) port of the Discovery board to drive the SSD1963. 

> DMA, have tutorial how to config ? (my case Cortex-M7 ).

You need a "memory to memory" DMA transfer with FSMC because the parallel port is mapped as a RAM address range.  So it's possible to use DMA for the parallel port, but let's make the FSMC working first! :)

> Question 3: RAM start and size ?

I suppose you mean the external SRAM address. In this coniguration, the framebuffer is located in the SSD1963 so don't really need the external SRAM.

autoprog, 2018-12-30:
Hi, I've already tested the discovery card and everything is fine. But now I want to connect another RT1021 processor with the Lvgl library, used these shields, the problem is that I do not know how to do it, because they do not have the pins (LCD_TFT HSYNC | LCDTFT VSYNC), how does Lvgl works without these pins?.

see link: https://pt.aliexpress.com/item/7-7-0-Inch-TFT-LCD-Display-800x480-SSD1963-Touch-Panel-Screen-PWM-LED-Backlight-Controller/32879036398.html?spm=a2g03.10010108.1000016.1.31b95c083GlriH&isOrigTitle=true

thanks,
Carlos.

autoprog, 2018-12-31:
Hi, I found what I want to clarify, see this AN, it shows that the core does the work of time signals.
Lvgl, can you work with this configuration?

[image]

link AN: http://www.solomon-systech.com/files/ck/files/SSD1961_2_3_Application_note_v1.7.pdf

tnanks,
Carlos.

kisvegabor, 2018-12-31:
It's definitely possible! You just need to use the parallel port of your MCU. See this example driver: https://github.com/littlevgl/lv_drivers/blob/master/display/SSD1963.c

kisvegabor, 2019-01-03:
Just a minor thing: FSMC is the parallel port on STM and LTDC is for RGB interface

closed 2019-01-03 by kisvegabor