lvgl/lvgl #679: chrome tab style

ali-rostami, 2018-12-29:
Hi 
I want to implement a tab object wich its style is like chrome browser:
[image]
how can I make an out curve for tabs -like the bottom corner of Chrome's tab-?

kisvegabor, 2018-12-31:
Hi,

I made some experiments to implement this but all my tries looked ugly. I'd like to suggest to use images instead and just saw your update. 

As the shape of the tabs is quite complex I think using images it the best way to do it. 

Can you show how your solution looks like?

embeddedt, 2018-12-31:
@ali-rostami @kisvegabor When you are using the images in a tabview would you have a `set_text` API to draw text on top or would you have to add the text in the image itself?

kisvegabor, 2018-12-31:
@embeddedt I didn't use images just tried to adjust the styles but the result didn't look well.

I think @ali-rostami didn't use the tabview object but created something custom. But we will see when he explains.

ali-rostami, 2019-01-01:
Yep, I didn't use tabview and instead put 4 imgbtn beside each other.
I will upload a picture of the result as soon as posible.

kisvegabor, 2019-01-01:
I see, thank you for sharing!

ali-rostami, 2019-01-01:
you're welcome.
Isn't it possible to use imgbtn in tabview instead of btnm? or doing something similar, you know, let us put a picture for tabs in pressed or released states.

kisvegabor, 2019-01-01:
`btnm` draws rectangles which cannot be converted to images. I can't imagine a good way to achieve it with `lv_btnm`. :(

kisvegabor, 2019-01-03:
@embeddedt And the images would be drawn instead of the buttons?

embeddedt, 2019-01-03:
@kisvegabor The images would be drawn first, then the text would be drawn on top. Basically we could have the image replace the currently drawn button background if specified. Does that make sense?

ali-rostami, 2019-01-03:
I think we can have imgbtnm object and then use this new object -instead of btnm- in making the new tabview object kind.

embeddedt, 2019-01-03:
@ali-rostami Wouldn't creating `imgbtnm` be the same as extending `btnm` with image functionality?

ali-rostami, 2019-01-03:
Yes, but as you know imgbtn can get different images for their diffrent states. 
I think we need some thing like this for the new object.

embeddedt, 2019-01-03:
@ali-rostami That functionality could also be added to `btnm` via styles.

ali-rostami, 2019-01-03:
Yes, we could do this for imgbtn too.
I mean instead of defining this new object -imgbtn- we could add this functionality to btn via styles, couldn't we?

embeddedt, 2019-01-03:
@ali-rostami I am not exactly sure why the image button object was made as a separate object. @kisvegabor should know.

If we did add image functionality to `lv_btn`, we would have to keep `imgbtn` around till at least 6.0 for backwards compatibility.

kisvegabor, 2019-01-04:
@embeddedt I created `imgbtn` as separate object because of two reasons:
1. **Memory usage** It needs to store `const void * img_src[LV_BTN_STATE_NUM]; ` -> 4 * 5 byes
2. **Limited size** You can't set the size of an image button. `LV_IMGBTN_TILED` makes possible to set the width at least by repeating a center image.

I see it would be great to add image support to button matrix but the size of the images can't be set as the size of the button according to the size of the button matrix itself.

Or do you think it's not a problem? Just align the images to center?

embeddedt, 2019-01-04:
@kisvegabor Every other GUI with an image widget that I have seen either resizes the widget to fit the image, or aligns the image to the center. So I think aligning to the center is an acceptable solution.

kisvegabor, 2019-01-04:
@embeddedt 
I see. If center aligned images can work then we can do this. 
Just need to find a way to efficiently store the image source pointers. I mean if we add 5 image sources for the 5 possible states we need to assign them to buttons. In the case of @ali-rostami if we add the tab images we should tell that all the buttons use it.

Maybe we can extend the control byte in the button map to have an id for the image sources. Or do you have any idea how to do it?

kisvegabor, 2019-01-05:
@embeddedt 
Only 1 right now. But in `v6.0` we can improve it.

embeddedt, 2019-01-05:
@kisvegabor I see. In that case we would have to wait till `v6.0` to implement this feature because there are no free bits in the control byte right now.

kisvegabor, 2019-01-05:
@embeddedt 
Great!

@ali-rostami 
Do you also like it?

ali-rostami, 2019-01-06:
wow, there are some things that I don't understand.
what do you refer to by saying control byte(s)?
why can't we have more than 1 byte for control byte right now?

> @embeddedt
> Only if you like the idea of handling it via the control byte(s).

Now I'm confused. why should we wait for ver 6.0, if we can do it now?

PS1: 
I'm so sorry If these are simple questions, but to be honest with you I need to know these to answer you dear @kisvegabor and @embeddedt :) thanks for your patience.
PS2:
Just for notice: the picture of the most left tab is different from the most right tab and these two are different from all other tabs. so I think that we can't apply all of the tabs the same picture.

ali-rostami, 2019-01-07:
Now it's more clear for me. 
@kisvegabor  Sounds good.

kisvegabor, 2019-04-10:
@ali-rostami 
I was thinking about this feature again. I'm not sure if it's really a good idea to mix button matrix with images because it would break the responsive nature of the button matrix.

If you are still interested in these features we can talk about it and discuss the details to see how it could work.

ali-rostami, 2019-04-10:
I solved my problem by putting 4 imgbtn beside each other. so I don't need it anymore and not sure if there is a real need to this feature. 
maybe it's better to wait for the future to see if anyone has a situation that makes implementing this feature reasonable. what do you think?

kisvegabor, 2019-04-11:
I agree let's wait to see if there would be requests on it.

I close until that.

closed 2019-04-11 by kisvegabor