lvgl/lvgl #681: More inconsistencies between .c and .h files

rreilink, 2018-12-29:
In succession of issue #660, [my script](https://github.com/rreilink/pylvgl/blob/unify_with_mpy/check_decls.py) found the following inconsistencies between function declarations (.h files) and definitions (.c files):

```
lvgl/lv_core/lv_obj.c
Not previously declared: lv_opa_t lv_obj_get_opa_scale_enable(const lv_obj_t *obj)

lvgl/lv_misc/lv_txt.c
   uint16_t lv_txt_get_next_line(const char *txt, const lv_font_t *font_p, lv_coord_t letter_space, lv_coord_t max_l, lv_txt_flag_t flag)
   uint16_t lv_txt_get_next_line(const char *txt, const lv_font_t *font, lv_coord_t letter_space, lv_coord_t max_width, lv_txt_flag_t flag)
   static uint32_t lv_txt_utf8_prev(const char *txt, uint32_t *i_start)
   static uint32_t lv_txt_utf8_prev(const char *txt, uint32_t *i)

lvgl/lv_misc/lv_ll.c
   void lv_ll_init(lv_ll_t *ll_p, uint32_t node_size)
   void lv_ll_init(lv_ll_t *ll_p, uint32_t n_size)
Not previously declared: void lv_ll_swap(lv_ll_t *ll_p, void *n1_p, void *n2_p)

lvgl/lv_misc/lv_circ.c
Not previously declared: void lv_circ_init(lv_point_t *c, lv_coord_t *tmp, lv_coord_t radius)
Not previously declared: bool lv_circ_cont(lv_point_t *c)
Not previously declared: void lv_circ_next(lv_point_t *c, lv_coord_t *tmp)

lvgl/lv_draw/lv_draw_line.c
   static void line_draw_hor(line_draw_t *main_line, const lv_area_t *mask, const lv_style_t *style, lv_opa_t opa_scale)
   static void line_draw_hor(line_draw_t *line, const lv_area_t *mask, const lv_style_t *style, lv_opa_t opa_scale)
   static void line_draw_ver(line_draw_t *main_line, const lv_area_t *mask, const lv_style_t *style, lv_opa_t opa_scale)
   static void line_draw_ver(line_draw_t *line, const lv_area_t *mask, const lv_style_t *style, lv_opa_t opa_scale)

lvgl/lv_objx/lv_page.c
   static void lv_page_sb_refresh(lv_obj_t *main)
   static void lv_page_sb_refresh(lv_obj_t *page)

lvgl/lv_objx/lv_ta.c
   static bool lv_ta_design(lv_obj_t *ta, const lv_area_t *mask, lv_design_mode_t mode)
   static bool lv_ta_design(lv_obj_t *ta, const lv_area_t *masp, lv_design_mode_t mode)

lvgl/lv_objx/lv_btn.c
   static bool lv_btn_design(lv_obj_t *ddlist, const lv_area_t *mask, lv_design_mode_t mode)
   static bool lv_btn_design(lv_obj_t *btn, const lv_area_t *mask, lv_design_mode_t mode)

lvgl/lv_objx/lv_ddlist.c
   void lv_ddlist_open(lv_obj_t *ddlist, bool anim)
   void lv_ddlist_open(lv_obj_t *ddlist, bool anim_en)
   void lv_ddlist_close(lv_obj_t *ddlist, bool anim)
   void lv_ddlist_close(lv_obj_t *ddlist, bool anim_en)

The following declared functions are not defined:
void lv_vdraw_refresh_temp_arrays(void)
```

kisvegabor, 2018-12-30:
Thank you, I will fix them soon!

kisvegabor, 2019-01-01:
I've fixed the listed issues in `dev-5.3`.

The only interesting thing was with `lv_circ` because those functions were declared in `lv_circ.h` but `lv_circ.h` was not included in `lv_circ.c`. I included it, may only it caused the issue.

closed 2019-01-10 by kisvegabor